esp8266/Arduino #3133: ESP_12 SPI with ublox m8n

rt3j, 2017-04-11:
Hello everyone, I would like to connect my GNSS of virtual robotix that has a neo m8n ublox GNSS receiver with my esp 12 via SPI.

I set the GNSS module in SPI mode via UART... done. But I cannot find anything relevant to use SPI on esp 12. I saw some functions on the manual but those was just to connect 2 esp 12 together. I can't find any API or example how to use an SPI in esp_12. Can you help me? 

I just would like something similar to the serial like: SPI.read(buffer_read); and SPI.write_to_slave(string);

ecc....    thanks!

jjssoftware, 2017-04-13:
I've not done this myself so this is the best I can offer:

* Mentioned here: https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#spi
* The SPI library code is included in this project here: https://github.com/esp8266/Arduino/tree/master/libraries/SPI
* This is an ESP8266 flavour of the original arduino SPI library which has documentation here so I should hope that this documentation is fully valid: https://www.arduino.cc/en/reference/SPI
* Examples are here: https://www.arduino.cc/en/Tutorial/BarometricPressureSensor and here: https://www.arduino.cc/en/Tutorial/DigitalPotControl

rt3j, 2017-04-18:
Ok but my problem is that I'm not using arduino, I'm using the SDK and the GCC compiler on linux and I'm compiling on the linuz terminal. It's the first time that I don't use a normal IDE like arduino, this is the fact a can't find anything but Thanks for the answer anyway, I'll try to reassembly all this code to use them in my code :) ....

jjssoftware, 2017-04-18:
ah ok I probably made an incorrect assumption you were building stuff with the esp8266 arduino core (this project) since the issue was raised here.

Good luck, sounds like a challenge 😀

rt3j, 2017-04-19:
pff yes sorry, it's my fault.. anyway I think I'll transfer my project on arduino....

closed 2017-04-19 by rt3j