esp8266/Arduino #3134: ESP8266 Wireless "pm open,type:2 0" Problem...

cbrum11, 2017-04-11:
## INFO

Using a CO2 sensor with an ESP8266 via I2C.  Transmitting data over wifi to a raspberry pi access point via MQTT Mosquitto broker.  Works great for anywhere between 3 to 9 readings but then I get an error (or what I assume is an error)...

**"pm open,type:2 0"**

I assume this is the error because the readings always stop on the next set after this bugger shows up.
I have no idea what it means, so any suggestions would be GREATLY appreciated.

Below is both the code and debug output.  Originally I removed all MQTT and Wireless parts from the code to narrow down the problem.  When all of this was removed the readings were perfect for hours.  Then I removed only the MQTT parts and left the wireless.  BOOM... readings stop after a few iterations... the problem is somewhere in the wireless, not with MQTT.  In addition to the pm open error, there also seem to be some weird reconnects (from debug) happening when the connection is first made.  Are these supposed to occur?

## DEBUG
```
The Greaux Engineering CO2 System is Active...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
state: 5 -> 2 (2c0)
rm 0
Initializing the Sensor...
reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with greauxeng, channel 7
dhcp client start...
ip:192.168.0.11,mask:255.255.255.0,gw:192.168.0.1
Sensor Initialized

Connecting to greauxeng
state: 5 -> 0 (0)
rm 0
del if0
usl
mode : null
mode : sta(5c:cf:7f:10:4e:5a)
add if0
f r0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
state: 5 -> 2 (2c0)
rm 0
..reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with greauxeng, channel 7
dhcp client start...
.ip:192.168.0.11,mask:255.255.255.0,gw:192.168.0.1
.
[... 279 lines cut ...]

      else if ((CO2_3 > CO2min) && (CO2_3 < CO2max)) {
        Serial.println("CO2 is IN RANGE");
        Serial.println(CO2_3);
        delay(1000);
      }

      else {
        Serial.println("CO2 is HIGH");
        Serial.println(CO2_3);
        tone(BuzzerPin, 2000, 10000);
        delay(1000);
      }
    }
    else
    {
      Serial.println("Reading is NOT ACCURATE. BEGINNING NEW READINGS...");
      Serial.println(CO2_3);
      delay(1000);
    }
  }
  delay(2000);
}

```

cbrum11, 2017-04-11:
## Additional Information

If I add the below to the beginning of the loop

```
 Serial.println("Checking WiFi Status");
  WiFi.disconnect();
  if ( WiFi.status() != WL_CONNECTED ) {
      delay ( 250 );
      setup_wifi();
  }
```
The code still drops out after 3 to 6 readings...  After this, the readings stay zero even after the Wifi reconnects successfully.  Once the readings become 0, the only way to fix it is to power cycle the Wemos...

I have also tried adding the following after reading some other posts... but this did nothing.

At the top of the sketch:
```
extern "C"{
#include "user_interface.h"
}
```
In the setup_wifi() function

`wifi_set_sleep_type(NONE_SLEEP_T);`

After doing this the debug error changed to pm open,type:**0** 0
and the line **sleep disabled** appeared, so I know the command took...

Any ideas....?

Thanks,
Chase

cbrum11, 2017-04-12:
## Update

I got the below code to work for over an hour so far with the wireless enabled.  The problem now, however, is when I add the MQTT, the output, eventually, reverts back to 0 readings.

All I've done is play a bit with the delay values and added:

`Serial.printf("Connection status: %d\n", WiFi.status());`

along the previously mentioned sleep and user_interface.h commands.  (I placed it in the main loop and the status always returns a 3 which means connected).

I'm not sure if the RAM is filling up and simply calling on the WiFi.status gives it a change to catch-up or what?  But it's a big bummer that as soon as I add the MQTT code back in, the original problem re-appears.  At least this feels like some progress...

I'll keep at this, but any help is GREATLY appreciated.  I'm in over my head here.

The below code works with wireless but **NOT** with MQTT

cbrum11, 2017-04-14:
I'm going to close this since I no longer believe this is an ESP8266 issue.  Currently I have hooked the CO2 sensor up to a relay and every-time I get a zero reading, I power cycle the sensor.  This solves the problem but is a pretty crude work-around.  I'm not sure if the sensor is going to sleep or if the issue is with the ESP8266, but I believe it is more likely sensor related.  It is strange however, that the sensor malfunction appears to be MQTT code related and does not happen without MQTT code... Anyway,  I'll just use my current work-around for now.

closed 2017-04-14 by cbrum11

ThomasBj, 2018-02-26:
I experience exaclty the same problem. Adding MQTT code halts execution completely.
I am using WifiMulti.
Here is the debug code i receive

--------------WiFi no longer connected!
--------------WiFi no longer connected!
--------------WiFi no longer connected!
--------------WiFi no longer connected!
scandone
[WIFI] scan done
[WIFI] 8 networks found
 --->  0: [4][10:62:EB:F2:40:24] ELLA_24GHz (-61) *
       1: [4][0C:96:BF:85:50:1A] Tele2Internet-2850A (-91) *
       2: [6][4C:32:75:C5:68:8A] Pettersson (-95) *
       3: [6][88:41:FC:AB:D7:23] Telia-16797D (-92) *
       4: [6][4E:8A:68:C5:75:30] Olas (-94) *
       5: [11][E0:B9:E5:E4:86:69] Pettersson 2,4 (-92) *
 --->  6: [11][10:13:31:6B:9C:3B] TN_24GHz_6B9C3B (-78) *
       7: [11][58:98:35:7B:8D:8D] TN_private_7B8D8D (-91) *

[WIFI] Connecting BSSID: 10:62:EB:F2:40:24 SSID: ELLA_24GHz Channal: 4 (-61)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with ELLA_24GHz, channel 4
dhcp client start...
wifi evt: 0
pm open,type:2 0

Rob58329, 2023-11-14:
Re. ESP8266 “pm open,type:2 0” wifi issue.

I understand that this usually occurs during a wifi connect attempt, after the “connected with ????, channel 1” and “dhcp client start” events (and is apparently supposed to mean something has entered a sleep state).  It immediately causes a “state: 5 -> 0 (0)” and the ESP8266 abandons the wifi connection attempt.  (At it had already “connected with ????...” it does not do any subsequent “no ???? found, reconnect after 1s” activities.)

I believe it is being caused by an error in the internal (closed source) client-ESP8266 code which gets the DHCP address from the remote router, as when it works you see “ip:192.168.?.100,mask:255.255.255.0,gw:192.168.?.1” instead of “pm open,type:2 0”.

Note that the error (for me) is a permanent one.  IE. if the client-ESP8266’s software is in a state where the error occurs, it will occur every time.  No amount of clearing the flash/wifi settings and re-flashing of the same software will fix it.

Some users have said that changing the remote router has helped.

In my case the remote router I am trying to connect to is a 2nd router-ESP8266 (in “WiFi.softAP()” mode).

And in my case the issue appears to have something to do with my ESP8266 code, as if I change the code in EITHER the client OR the router, I can get it to always work perfectly again.

So I believe it is an issue related to failing to get a DHCP address, possibly caused by some weird timing-conflict (between the code running on the client-ESP8266, and the code running on the remote router).

To fix it I just need to change my code in the client-ESP8266 unit and I can change it from never working to always working.  And it usually only needs a very small change such as adding a new “Serial.println()” line in “setup()”, which would seem to suggest there might also be a code-alignment/word-alignment issue somewhere.

Finally note that even when I’m getting the “pm open,type:2 0” issue between one “client-ESP8266 (eg. AT Tinker branded)” and one of my “router-ESP8266”:
* a different-client-ESP8266 (SAME brand and with same software) will still FAIL to connect to this “problematic-router-ESP8266”
* a different-client-ESP8266 (different brand but with same software) will connect fine to this “problematic-router-ESP8266”
* AND this “problematic-client-ESP8266” will still connect perfectly to a different-router-ESP8266.

Which would seem to imply that it is also effected by the hardware it is running on.  Note that the same code will work on some of my ESP8266s, but NOT on other of my ESP8266s.  (All my ESP8266 are D1mini’s, but purchased at different times with different branding.)

**Summary:**
I believe (my best guess) that some ESP8266-brands are effected by a timing (or maybe overflow memory location) issue in the closed-source code when obtaining a DHCP address from a remote router, which depending on your software alignment can cause a “pm open,type:2 0” error which stops the wifi connecting.  But this can be fixed by adjusting your software slightly (either on client OR changing router software/hardware), or changing to a different ESP8266 unit.

I hope this info will be of use.

d-a-v, 2023-11-14:
This is dark magic :)
`pm open` is `power management` and it generally does not prevent ongoing events.
With one of your failing board, you can try
- erase all flash once
- force 802.11g
- reduce power management