esp8266/Arduino #3136: esp8266 sending data to eachother

kasraeee, 2017-04-14:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-07

I am using 2 esp07 modules , one is configured as an access point and it is a server(host)  which listens to port 23 and it has a static IP .
The other one is a client which connects to the access point  in the setup  , and in the loop , Whenever it needs to send data, it  connects to the server(host) IP and port 
client.connect(host, port);
client.print(data);
and then disconnects after data is sent . 
client.stop();
This was very good for me and worked well before i decided to send some data from the host to the client as well. As you know when i use   client.stop(); , the host can not send data to to client since the connection is closed . 
If I don't close the connection after the data is sent , I won't be able to use client.connect(host, port); function to connect to the host again . On the other hand if I use client.connect(host, port); only once  in setup , then if somehow the connection is closed accidentally (as i have faced this problem sometiomes in the past ) I can not understand it to connect again.
is there any function to check whether the connection to host (not the access point ) is still running ?
does any one have any solution to send data bidirectionally between host and client in a proper way ?
best regards

jjssoftware, 2017-04-14:
Take a look at:

https://github.com/esp8266/Arduino/tree/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFiMesh

You might be able to just use ESP8266WiFiMesh as a drop in replacement for whatever you have at the moment. Or just take a look at the source, it might give you some ideas.

kasraeee, 2017-04-14:
Thanks alot JJ for your suggestion 
I would like to know when esp disconnects from the host since sometimes my host is a software and i must manage several esp modules by a software on my pc . i doubt whether i can perform mesh on the software too , 
by the way i took a look at  the code, unfortunately I'm not so professional to understand the wifi mesh codes completely and use it the way i need .

jjssoftware, 2017-04-14:
This seems to be a scope creep alert comment. It appeared as though you just had ESP devices talking to each other but now we seem to have use of some PC software in the mix..

In the ESP mesh network, if there was a requirement to know when a client disconnects from network and if it's an elegant disconnect, then a client could send a message just before it disconnects to notify interested hosts in the mesh that it's going away. 

Theoretically when a message is sent in the mesh, the ESP8266WiFiMesh doesn't really care about clients that have gone away because it determines eligible available networks with a WiFi.scanNetworks() call (see https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp#L128)

If you really do need some way of determining ESP hosts on the network outside of the mesh (for instance in PC software), you would need some other method to determine what hosts are on the network. I guess there are numerous ways of doing this but there's just not enough info to go on to make any reasonable suggestions at the moment.

kasraeee, 2017-04-14:
"This seems to be a scope creep alert comment"
no I just wanted to say that it is better for me to find a solution besides wifi mesh , what i meant by " sometimes my host is a software " was not for the recent project .
In fact what is really performed is this : 
1-The esp module which is in access point mode (and it is the host too) has a relay to control a gate . it opens the gate whenever its sensor detects someone entering the gate in outside - inside direction. 
2- the second esp which is our client has a sensor . when sensor detects someone entering in the opposite direction (inside-outside) it sends a string to the host to open the gate . 
all the above was working fine as described in my first comment until the customer asked me to ad RTC to this projectc  and log these transactions on a SD card .
 Since I  space problem as well as pins , I could just add SD socket to the first board and add ds3231  RTC to the second board .

jjssoftware, 2017-04-14:
Ok no worries. From what you've described it does seem that bi-directional comms over a mesh based on ESP8266WiFiMesh would still work for you. That is unless there's some physical hardware or external interference reason for the communication breaking down between the ESP devices..

But I guess you know your application best so if you're against the mesh idea or any the ideas implemented in that code no problem, I'm not here to hard sell it to you or anyone else ;)

devyte, 2017-09-07:
@kasraeee please post full minimal sketches of what you are trying to accomplish. Also, please edit your original post and add the additional fields found in the issue template (you can see the template when opening a new issue).