esp8266/Arduino #3137: ESPhttpUpdate: weird problem with ESP.restart()

neilyoung, 2017-04-14:
I'm using literally the same procedures as suggested in the example httpUpdate.ino.

The HTTP_UPDATE_OK return is never coming, what is understandable, because the ESPhttpUpdate restarts the ESP before. In this case I don't get back control and the ESP restarts, but it takes say > 5 secs until the restart is done. After that the new version is running.

If I'm calling 

 `ESPhttpUpdate.rebootOnUpdate(false);`

in order to do ESP.restart() by myself as a reaction to HTTP_UPDATE_OK, then the restart happens very quickly, but the version is not updated.

I'm pretty baffled: I cannot see a major difference doing the ESP restart with the ESPhttpUpdate module compared to doing it outside. Any pointers welcome

jjssoftware, 2017-04-15:
It seems odd that calling `ESPhttpUpdate.rebootOnUpdate(false);` like this results in a device restart at all. The parameter being passed as false here is the reboot parameter and this controls whether the device should restart following the update having been applied. By passing false, you're telling the device to not restart.

* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L69
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp#L300

neilyoung, 2017-04-15:
@jjssoftware Thanks for the quick response. 

Sorry, maybe I was not specific enough. 

`ESPhttpUpdate.rebootOnUpdate(false);
`
does _not_ lead to a restart. I was inserting this line, because I saw, that the update process didn't return to me with HTTP_UPDATE_OK on OK. After having a look into the source code of the updater I found it reasonable, because the update process implicitly restarts before returning, if rebootOnUpdate is true.

Hence I thought, telling the updater to not restart and do the ESP.restart() on reception of HTTP_UPDATE_OK would be exactly the same.

The thing is: It is not the same and I have no explanation for this. If I don't let the updater restart and do the restart by myself, the firmware is NOT updated. The restart looks exactly the same in both cases, just the "internal" restarts lasts ways longer and after that I have my new version "on disk".

I was intending to clean up things on HTTP_UPDATE_OK and store some information into RTC Ram in order to know next time.

I can work around, but somehow I don't get it :)

neilyoung, 2017-04-15:
BTW: Nice to see two monkeys talking :)

jjssoftware, 2017-04-15:
Ah right I understand now, I agree this is odd and I also agree with your thinking that you should be able to call ESP.restart() yourself and achieve the same effect - newly flashed firmware.

I'm not sure how useful this is going to be but I'll mention it anyway; I don't use the ESP8266HTTPUpdate class myself for OTA but I do use the UpdaterClass directly along with a HTTP uploaded firmware file. I know that ESP8266HTTPUpdate also uses the UpdaterClass internally to validate and perform the update so there is commonality here. I manually call ESP.restart() following Update.end() being called and it does work fine, but one thing to note is that I delay calling restart by 1000 msecs after the update is applied and call it in the loop. I built this stuff over a year ago so unfortunately I can't remember if I encountered the same thing you're seeing now..I guess I need more bananas :)

neilyoung, 2017-04-15:
:) I already tried to delay(1000) before ESP.restart() to no avail. No other idea yet... An Easter egg somehow

devyte, 2017-04-15:
Hi, I seem to remember reading somewhere that the OTA update process works
like this:
1. Write firmware to flash as it is received. Location written is after
current firmware end and before SPIFFS begin.
2. Reboot
3. On start, overwrite old firmware with new firmware
4. Boot up from new firmware

If the above is true,  step 3 would explain a longer bootup time.

or mute
the thread

davisonja, 2017-04-15:
That's an accurate description of the OTA update process. Step 3 does
indeed take longer.

neilyoung, 2017-04-16:
I cannot confirm to see two reboots. If the restart is done internally the process is like this:

The question is: Why am I unable to achieve the same behaviour, if I 

`ESPhttpUpdate.rebootOnUpdate(false);
`

Do I have to call an additional function after having received HTTP_UPDATE_OK

neilyoung, 2017-04-16:
Having a look into the source, there really seems to be not a big magic https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp

The script is downloading the file and then calling Updater in order to flash the stuff. When this succeeds it call ESP.restart() if reboot is true and !spiffs (spiffs is false in my case). So the call to 

`t_httpUpdate_return ret = ESPhttpUpdate.update(url);`

does not return in positive case. This is what I'm seeing as well. 

If I set

`ESPhttpUpdate.rebootOnUpdate(false);`

then the only function called insidte ESPhttpUpdate is http.end() and then it returns HTTP_UPDATE_OK, so returns control to me. My ESP.restart() fails then... No idea why.

jjssoftware, 2017-04-16:
I still have no actual answer why this doesn't work when it looks so simple and looks like it should work..but I thought I'd take a tangential approach and go look at some other project that implements OTA update using the arduino core and ESPhttpUpdate.

One of my favourites is the Sonoff-Tasmota project and when I looked at that OTA code, I found that it does something very similar to what I do in my own code following firmware flash - the reboot is delayed in the loop (actually a state machine pattern in the Sonoff-Tasmota code) and not performed immediately. So I'm starting to think this is significant to getting this working..even if we don't know why this method works and calling ESP.restart() immediately and inline doesn't.

For reference:
* https://github.com/arendst/Sonoff-Tasmota/blob/824368125b80b95abb6c4cc66e4b8dbe824273fc/sonoff/sonoff.ino#L1562
* https://github.com/arendst/Sonoff-Tasmota/blob/824368125b80b95abb6c4cc66e4b8dbe824273fc/sonoff/sonoff.ino#L1845
* https://github.com/arendst/Sonoff-Tasmota/blob/824368125b80b95abb6c4cc66e4b8dbe824273fc/sonoff/sonoff.ino#L1857
* https://github.com/arendst/Sonoff-Tasmota/blob/824368125b80b95abb6c4cc66e4b8dbe824273fc/sonoff/sonoff.ino#L1889

devyte, 2017-09-06:
What happens if you try the last comment of #1722?

neilyoung, 2017-09-06:
@devyte Sorry, it is a long time ago, the project is closed and works. I think I found a way around, even though I don't remember, how :)

Thanks for the follow up.

devyte, 2017-09-07:
@neilyoung ok, I'll keep it open for now, but please report your findings back here. Also, please try with latest git.

neilyoung, 2017-09-07:
@devyte I have closed that issue now. Thanks

closed 2017-09-07 by neilyoung

lucasCFO, 2018-09-27:
Hello, I have the same problem and it seems you have found the answer but you didn't report it... Can you explain me what's the trick pls ??

accuware, 2018-09-27:
@lucasCFO Well, if I recall correctly, I could not set `ESPhttpUpdate.rebootOnUpdate(false);` and do my own ESP_restart() on HTTP_UPDATE_OK. I think, I'm now relying on the internal restart performed by the ESPhttpUpdate instance, which works for me.

I checked my sources and found this

```
                // Does not work
                // ESPhttpUpdate.rebootOnUpdate(false);

```

neilyoung, 2018-09-27:
@lucasCFO  Sorry, used my business account for the previous answer...

matijagaspar, 2019-06-22:
If anyone stumbles upon this issue: the update stores some info into the rtc memory (first 32 blocks), so be extra sure that if you use rtc memory to leave those first 32 blocks free. Set offset 33 (this is 33*4 bytes) of your writes.

This solution helped me get it to work correctly.