esp8266/Arduino #3138: Fatal exception (0)

hulr, 2017-04-15:
Hello everyone, here my problem description:

### Hardware
**Hardware:**			NodeMCU v2 Dev Board
**Core Version:**      	2.3.0

### Description
I have two exact same setups including:

- Node MCU v2 Dev Board
- Power supply: DC 5V 1A (same model)
- USB 2.0 <-> Micro USB cable (same type)

The upload of the same sketch works on both boards without any problems and the first board is just working fine, but when I try to boot the second board, I receive the messages stated in the "Debug Messages" section. 
After mutliple tests with example sketches (e.g. ESP8266 Blink) and flashing the original NodeMCU firmware I still have no luck. 
I think it's a hardware malfunction or do you have other possible solutions?

### Settings in IDE
**Board:**  			NodeMCU 1.0 (ESP-12E Module)
**CPU Frequency:**        80MHz
**Flash Size:** 		4M (3M SPIFFS)
**Upload Speed:**	        115200
**Upload Using:**		Serial

### Debug Messages
```
00000000, epc3=0x00000000, excvq⸮m⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc9f0c112
~ld
Fatal exception (0): 
epc1=0x40210bf4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210bf4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40210bf4, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
decoded with EspExceptionDecoder:
```
0x40210bf4: scan_parse_beacon at ?? line ?
```

tttapa, 2017-04-22:
Could be a flash memory failure.
Exception (0) indicates an illegal instruction.
I sometimes get this when uploading at speeds > 115200 baud, but reuploading always solves the problem.

Pieter

hulr, 2017-04-22:
Thank you for your comment.

I tried to reupload the compiled sketch using a baud rate of 9600. After a failed attempt with the Arduino IDE (same fatal exception message 0), I used the esptool and got other exception messages. It's funny that I'm receiving different exception messages on each reset:

**1st reset**
```
Fatal exception 0(IllegalInstructionCause):
epc1=0x402199e8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception 28(LoadProhibitedCause):
epc1=0x401061da, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000300, depc=0x00000000
```

**2nd reset**
```
Fatal exception 0(IllegalInstructionCause):
Fatal exception 9(LoadStoreAlignmentCause)
```

**3rd reset**
```
0x402199b7: rijndaelKeySetupEnc at ?? line ?
0x40219922: rijndaelKeySetupEnc at ?? line ?
0x40224f7c: sys_timeouts_init at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 272
0x40219e24: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105294: wdt_feed at ?? line ?
0x40219e0e: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105570: wdt_feed at ?? line ?

```

**4th reset**
```
0x40106058: system_restart_core at ?? line ?
0x4021af53: system_restart_local at ?? line ?
0x4021ae93: system_restart_local at ?? line ?
0x40201f21: __wrap_system_restart_local at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x40201f37: __wrap_system_restart_local at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201f34: __wrap_system_restart_local at C:\Users\USER\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x402138b8: ieee80211_send_action at ?? line ?
0x40231d80: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x40219c3f: user_uart_wait_tx_fifo_empty at ?? line ?
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x401052b3: wdt_feed at ?? line ?
0x40105294: wdt_feed at ?? line ?
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x401052b3: wdt_feed at ?? line ?
0x402199e8: rijndaelKeySetupEnc at ?? line ?
0x4021a70a: ets_vsprintf at ?? line ?
0x4021a733: ets_sprintf at ?? line ?
0x402298f2: chip_v6_rf_init at ?? line ?
0x4021ccee: wifi_station_set_default_hostname at ?? line ?
0x4021ccd9: wifi_station_set_default_hostname at ?? line ?
0x4021ccc7: wifi_station_set_default_hostname at ?? line ?
0x40224e35: sys_timeout at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 296
0x402199b7: rijndaelKeySetupEnc at ?? line ?
0x40219922: rijndaelKeySetupEnc at ?? line ?
0x40224f7c: sys_timeouts_init at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 272
0x40219e24: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105294: wdt_feed at ?? line ?
0x40219e0e: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105570: wdt_feed at ?? line ?
```

**erase_flash**
```
esptool.py --port COM3 erase_flash
```
```
esptool.py v1.3
Connecting....
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 10.1 seconds
```

**write_flash**
```
esptool.py --port COM3 --baud 9600 write_flash -fm dio 0x00000000 app.bin
```
```
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0240
Wrote 241664 bytes at 0x0 in 251.8 seconds (7.7 kbit/s)...
Leaving...
```

lrmoreno007, 2017-04-22:
Surely there is some problem that resets the board. Once uploaded the program, try to bypass the basic pins for good operation with a wire.
GPIO0-3V3
Reset - 3V3
GPIO15 - GND
CH_PD-3V3
It happened to me with a NodeMCU that had the reset pull up badly I solved it by removing it and putting a new one of an old motherboard.

hulr, 2017-04-23:
I connected the pins as you mentioned, but it doesn't boot and I have no serial output.

D3 (GPIO0) - 3V3
RST - 3V3
D8 (GPIO15) - GND
EN - 3V3

lrmoreno007, 2017-04-24:
Try one by one to find the problem, because it seems that you must have a short-circuit in some and everything is coming down.

hulr, 2017-05-02:
I sent it back and got a new one, it was a hardware malfunction. Thanks for your help.

closed 2017-05-02 by hulr

juanitoddd, 2017-06-05:
This was happening to me aswell  (Reboot loop with all sorts of Exceptions) and when I was about to ditch the module, I deleted the whole flash,  re-uploaded my sketch and connected to a external 3.3v source and turn out to work !. I was using the 3.3v from my USB to TTL module. Apparently was causing my issue

shivamkeshri, 2017-12-12:
If you are getting fatal error exception like this:

Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000cbd9 depc=0x00000000
In infinite loop in your serial monitor of arduino IDE  .

then goto this link download the software and follow the procedure and erase the flash memory to solve the error.

This does not solve fatal error that occurs due to your program but in case your device goes in such condition that it can’t be able to access program memory then it will work and try atleast one time to solve the problem.

This is the procedure to hard reset the nodemcu  

( https://www.youtube.com/watch?v=MHrm7axsImI&t=146s  )

Step :
Install latest python version in you pc.(https://www.python.org/downloads )
Open cmd prompt as administrator .
Go to c/program files or program files (x86)->python (your version)->Script. For this type (cd c/program files (x86)/python(your version)/Script) then press enter .
Now type (pip install esptool).
Now download ESPlorer  ( https://esp8266.ru/esplorer/ ) version(Download ESPlorer.zip (v 0.2.0-rc6)) and extract the file and open executable jar file .
Now goto nodemcu firmware site (https://github.com/nodemcu/nodemcu-firmware/releases ) and from  download file (nodemcu_float_0.9.6-dev_20150704.bin ) and copy this file into the c/program files (x86)/python(your version)/Script folder . 
Now in cmd prompt just type.
	esptool.py --port COM(your port no.) --baud 115200 erase_flash
And press enter.

Note : you can see your port no. into the device manager .

han-nes, 2018-01-28:
@juanitoddd do you know the reason why 3.3v from the USB to TLL doesn't work? I tried to power up my Esp8266-03 using a voltage divider from 5v to 3.3v (47R and 1M) and it caused the same error...

devyte, 2018-01-28:
@han-nes you can't use a voltage divider to power the ESP. The current spikes cause voltage dips that result in resets and odd behavior.

PrimeTime416, 2019-01-07:
So true for me with the power supply! I have one of the USB to serial TTL FTDI Breakout Boards and there are jumper pins for setting the VCC(io) to 3.3v or 5v http://osepp.com/wp-content/uploads/2012/01/OSEPP_FTDI-Rev1.1-Schematic.pdf and unfortunately I had it set to 3.3v going to my Adafruit Feather Hazzah which has a 5v to 3.3v regulator https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/downloads, see the problem here! Anyhow so I  just set the jumper to 5v VCC position and my problem went away +1: thanks for the insight I would not have resolved this issue without this initial post, thanks folks :)

hmallen, 2019-07-10:
For the sake of completeness I'll add that I also had this problem, except it occurred for me when powering the NodeMCU with a 12V adapter at the same time as being plugged into USB. I saw the shottky diode on USB in the schematic and assumed it wouldn't be a problem, but it didn't seem to like it very much.

ramanraja, 2020-03-16:
In my experience, software problems have been the single most common cause of the error. It is difficult to debug your code without knowing anything about it, but general good practices that helped my team:
1. Liberally place debug statements throughout your code, and monitor the serial port. The idea  is to find out exactly after what point the MCU resets.
2. If you had a previous version of the code that was working without crashing, upload that code again. Then slowly add the new features one at a time and see what triggers the crash.
3. In general, keep developing the code a few lines at a time, and test it after every iteration. This will immediately point to the source of a crash as the latest changes in code.
4. Get a fresh NodeMCU/ ESP chip and load the code on it. If it works, the old chip is the suspect.
5. To check if Flash memory is corrupted, wipe out the memory from the Arduino IDE: select Tools/Erase Flash/All Flash Contents.
6. Before you do anything else, change the power supply with a robust 5V / 1A supply adapter for Node MCU. The power supply provided by FTDI programmers are always weak.
7. If everything else fails, solder a large capacitor (like 1000 MFD) very close to the power supply pin of the ESP chip. It is always a good practice to include this capacitor as part of your design, even if you are only using a bread board.

tingww, 2020-04-16:
I got this problem on my nodeMCU 1.0 when working on my project with timer interrupt, solved it by adding ISR function to IRAM by adding keyword ICACHE_RAM_ATTR.

charithjperera, 2020-09-25:
I had the same error messages as I had an optoisolator between D4 (GPIO2) and GND through a 2.2k. This load needed to be moved to another pin as D4/GPIO2 needs to be high to boot as per https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

gerivega, 2022-01-30:
> 

Thanks! It worked
esptool.py --port COM(your port no.) --baud 115200 erase_flash

johnchitsaka1, 2022-05-20:
Thank you truely helpful