esp8266/Arduino #3139: repeated rst cause 4 errors in arduino serial monitor

saurabhvyas, 2017-04-15:
ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset
rl

I am using node mcu v3 , This is what I am getting in my serial console instead of server ip address , the code is below

#include <ESP8266WiFi.h>
 
const char* ssid = "hcker";
const char* password = "c18rajouri";
 
const int f1 = 10; // GPIO10
const int f2 = 5;

const int b1=9; // gpio 9

WiFiServer server(80);
 
void setup() {
  Serial.begin(115200);
  delay(10);
 
  pinMode(f1, OUTPUT);
  pinMode(f2, OUTPUT);
  pinMode(b1, OUTPUT);

  
  digitalWrite(f1, LOW);
  digitalWrite(f2, LOW);
  digitalWrite(b1, LOW);
  
 
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
 
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
 
  // Start the server
  server.begin();
  Serial.println("Server started");
 
  // Print the IP address
  Serial.print("Use this URL to connect: ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");
 
}
 
void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
 
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
 
  // Read the first line of the request
  String request = client.readStringUntil('\r');
  Serial.println(request);
  client.flush();
 
  // Match the request
 
  int value = LOW;
  if (request.indexOf("/LED=ON") != -1)  {
    digitalWrite(f1, HIGH);
    digitalWrite(f2, HIGH);
    value = HIGH;
  }
  if (request.indexOf("/LED=OFF") != -1)  {
    digitalWrite(f1, LOW);
    digitalWrite(f2, LOW);
    value = LOW;
  }
 
// Set ledPin according to the request
//digitalWrite(ledPin, value);
 
  // Return the response
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");
 
  client.print("Led pin is now: ");
 
  if(value == HIGH) {
    client.print("On");
  } else {
    client.print("Off");
  }
  client.println("<br><br>");
  client.println("<a href=\"/LED=ON\"\"><button>Turn On </button></a>");
  client.println("<a href=\"/LED=OFF\"\"><button>Turn Off </button></a><br />");  
  client.println("</html>");
 
  delay(1);
  Serial.println("Client disonnected");
  Serial.println("");
 
}
 

I am using ubuntu 16.05 64 bit and arduino 1.8.3

igrr, 2017-04-16:
GPIOs 9 and 10 are used to interface SPI flash, you can't use them to drive LEDs. Make sure you're not confusing GPIO pin numbers and pin labels on NodeMCU board (they don't match). If you want to use NodeMCU pin label numbers, use constants starting with D for pin numbers, i.e. D0, D1, etc. These are defined to map to the correct GPIO number.

saurabhvyas, 2017-04-16:
Thank you for your reply , Indeed I was confusing pin numbers , closing this .

closed 2017-04-16 by saurabhvyas