esp8266/Arduino #3147: Struct data scramble

Thijxx, 2017-04-17:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos
Data read from Arduino is different than read from ESP. Same code, different data.

#### Hardware
Hardware:		NodeMCU ESP
Core Version:      	?

### Description
When I receive a struct via RadioHead library, the data is a mess when I print it from ESP and all good when I print in from Arduino, only change is pin number receiver.

Problem description
ESP struct is not Arduino struct

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		?qio?
Flash Frequency:	?40Mhz?
Upload Using:		USB
Reset Method:		nodemcu

### Sketch

```#include <RH_ASK.h>
#include <SPI.h>

#define BAUD 2000
#define RX 2
#define TX 12

RH_ASK driver(BAUD, RX, TX);

struct data {
  int id;
  int pin;
  int value1;
  int value2;
} myData;

void setup()
{
    Serial.begin(9600);
    Serial.println("Starting..");
    if (!driver.init())
         Serial.println("init failed");
}

void loop()
{
    uint8_t buf[RH_ASK_MAX_MESSAGE_LEN];
    uint8_t buflen = sizeof(buf);

    if (driver.recv(buf, &buflen)) {
      memcpy(&myData, buf, buflen);
      
      Serial.print("ID: "); 
      Serial.println(myData.id);

      Serial.print("Pin: "); 
      Serial.println(myData.pin);

      Serial.print("Value1: "); 
      Serial.println(myData.value1);

      Serial.print("Value2: "); 
      Serial.println(myData.value2);
      
      Serial.println("");
    }
}
```

### Debug Messages
```
ID: 263145
Pin: 18284827
Value1: 0
Value2: 0
```

On Arduino
```
ID: 1001
Pin: 4
Value1: 367
Value2: 284
```

JamesGKent, 2017-04-18:
try specifying the size of your integers, there are differences between the arduino default int sizes.
eg use int8_t for 8 bit, int16_t for 16 bit etc. this way it should be the same struct regardless of which arduino it is compiled on.

Thijxx, 2017-04-18:
yes thank you this solves it!

JamesGKent, 2017-04-18:
if solved then please mark this as closed. thanks

commit 65e0d8b5 (2015-05-20): New editor on MacOSX: CMD+E uses selection to find next entry. Fixes #3147
closed 2017-04-18 by Thijxx