esp8266/Arduino #3149: Question about wemos d1 mini

PomeloNicky, 2017-04-18:
I use wemos d1 mini to drive the ws2812 LED strip, following this tutorial  
https://www.youtube.com/watch?v=7Dv70ci-MOw
All went well till I use the strand test code in Adafruit NeoPixel library just like the vedio told.But it doesn't work. The same code went perfectly well if use Arduino Uno.
I'll appreciate it if anybody can help me! THANK YOU!

pipi61, 2017-04-18:
Hi!
whats not works?
If connect PC on serial port, whats view?
Wifi name/password set correct? connect? ipaddress ok?
Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

PomeloNicky, 2017-04-18:
No. I don't talk about wifi.Just download a strand test code to d1 mini module and the matrix don't work.
Can you understand me?I'm from China.Bad English......

发自网易邮箱大师
On 04/18/2017 22:22, Istvan Puzsar wrote:

Hi!
whats not works?
If connect PC on serial port, whats view?
Wifi name/password set correct? connect? ipaddress ok?
Serial.print("Connected to ");
Serial.println(ssid);
Serial.print("IP address: ");
Serial.println(WiFi.localIP());

davisonja, 2017-04-18:
WS2812B's are lots of fun.
I've used some happily with an ESP-12E (NodeMCU, rather than D1 Mini, but
I've got a D1 Mini so I can try it on that if that would be useful) but I
used the NeoPixelBus library rather than the Adafruit one, so my experience
may not be identical to yours.
The one thing that had me stuck was that I needed to set the pin to control
the LEDs to an output. I had assumed the library would do it, and it didn't
seem to.
Otherwise, so long as you're not trying to power too many LEDs from your
supply (if the D1 Mini is connected to a computer USB port, for example,
you might need to watch how many LEDs you're trying to turn on at once)

J,

devyte, 2017-09-06:
This is not the place to ask for help for your particular project. This is an issue tracker, meant for tracking issues in the core libs.
Please request assistance at esp8266.com or stackoverflow. If your project does lead to a bug in the core, please open a new issue, and fill in the fields in the issue template that you see when opening.
Closing.

closed 2017-09-06 by devyte