esp8266/Arduino #3150: WifiClient DNS lookup error after Wifi router is restarted

Anlo2846, 2017-04-19:
### Basic Infos

#### Hardware
Hardware:		NodeMCU v1.0 with ESP-12E
Core Version:      	2.3.0-rc2
Wifi router:		D-Link DIR-635

### Description

I'm using WifiClient to make HTTP GET requests at regular intervals. After a power outage I discovered that the ESP8266 had reconnected to the Wifi and was responding to connections, but no connections from the ESP8266 was possible. I have reduced the code to a minimum and can now reproduce the error by restarting the Wifi router.

After the Wifi router has been restarted and the ESP8266 is failing to connect, it still responds to pings. When millis() reaches 60000, ESP.restart() is executed and after booting it connects to the Wifi but still fails to connect to the web server. Removing power for 10 seconds is not enough, but removing power for ~15 minutes will make it connect again. Setting a static IP address instead of using DHCP will also make the problem go away.

### Settings in IDE

Module:  			NodeMCU v1.0 (ESP-12E Module), or Generic ESP8266 Module (for debugging)
Flash Size: 		4MB (1MB SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO (setting not available for NodeMCU)
Flash Frequency:	40Mhz (setting not available for NodeMCU)
Upload Using:		SERIAL
Reset Method:		nodemcu (setting not available for NodeMCU)

### Sketch

```cpp

#include <ESP8266WiFi.h>
unsigned long _lastHttpGet;

void setup()
{
  Serial.begin(115200);
  Serial.println("\r\n\r\nBoot");
  InitWifi();
}

void loop()
{
  if(WiFi.status() != WL_CONNECTED)
  {
    WiFi.disconnect();
    Serial.println("Wifi disconnected");
    InitWifi();
  }

  if(millis() - _lastHttpGet > 15000)
  {
    HttpGet();
    _lastHttpGet = millis();
  }

  if(millis() > 60000)
  {
    Serial.println("Restart");
    ESP.restart();
  }
}

void InitWifi()
{
  const char* ssid       = "Silence_of_the_LANs";
  const char* password   = "---";

  Serial.printf("Connecting to: %s", ssid);

  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while(WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.print("\r\nWifi connected: ");
  Serial.println(WiFi.localIP());
}

bool HttpGet()
{
  const char *url = "http://date.jsontest.com/";
  const char *hostname = "date.jsontest.com";
  const int port = 80;
  const int buflen = 1024;
  char buffer[1024];

  Serial.printf("Connecting to: %s:%d\r\n", hostname, port);
  WiFiClient client;
  if(!client.connect(hostname, port))
  {
    Serial.println("Connection failed");
    return false;
  }

  snprintf(buffer, buflen, "GET %s HTTP/1.1", url);
  Serial.println(buffer);
  client.println(buffer);
  client.println("Host: ESP8266");
  client.println("Connection: close");
  client.println();

  while(true)
  {
    if(client.available())
    {
      int n = client.readBytesUntil('\n', buffer, buflen - 1);
      buffer[n] = '\0';
      if(n > 0)
      {
        Serial.print("> ");
        Serial.println(buffer);
      }
    }

    if(!client.connected())
    {
      Serial.printf("Disconnected from %s\r\n", hostname);
      client.stop();
      return true;
    }
  }
}

```

### Debug Messages

#### Normal connect:
```
Boot
Connecting to: Silence_of_the_LANs.......
Wifi connected: 192.168.0.199
Connecting to: date.jsontest.com:80
GET http://date.jsontest.com/ HTTP/1.1
> HTTP/1.1 200 OK
> Access-Control-Allow-Origin: *
> Content-Type: application/json; charset=ISO-8859-1
> X-Cloud-Trace-Context: 37068d73f001c4f8aa64cdd0b1a63591
> Date: Wed, 19 Apr 2017 19:09:03 GMT
> Server: Google Frontend
> Content-Length: 100
> Connection: close
>
> {
>    "time": "07:09:03 PM",
>    "milliseconds_since_epoch": 1492628943567,
>    "date": "04-19-2017"
> }
Disconnected from date.jsontest.com
Connecting to: date.jsontest.com:80
GET http://date.jsontest.com/ HTTP/1.1
> HTTP/1.1 200 OK
> Access-Control-Allow-Origin: *
> Content-Type: application/json; charset=ISO-8859-1
> X-Cloud-Trace-Context: d42447f97978b65b8089ac96a8cb549d
> Date: Wed, 19 Apr 2017 19:09:19 GMT
> Server: Google Frontend
> Content-Length: 100
> Connection: close
>
> {
>    "time": "07:09:19 PM",
>    "milliseconds_since_epoch": 1492628959260,
>    "date": "04-19-2017"
> }
Disconnected from date.jsontest.com
```

#### Wifi router restart:
```
Wifi disconnected
Connecting to: Silence_of_the_LANs.............................
Wifi connected: 192.168.0.199
Connecting to: date.jsontest.com:80
Connection failed
Connecting to: date.jsontest.com:80
Connection failed
Restart

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
   ▒

Boot
Connecting to: Silence_of_the_LANs.......
Wifi connected: 192.168.0.199
Connecting to: date.jsontest.com:80
Connection failed
Connecting to: date.jsontest.com:80
Connection failed
Connecting to: date.jsontest.com:80
Connection failed
Restart
```

#### With Core+Wifi debugging enabled, normal connect:
```
Connecting to: date.jsontest.com:80
[hostByName] request IP for: date.jsontest.com
[hostByName] Host: date.jsontest.com IP: 173.194.221.121
:ref 1
GET http://date.jsontest.com/ HTTP/1.1
:wr
:sent 38
:ww
:wr
:sent 2
```

#### With Core+Wifi debugging enabled, after WIfi router restart:
```
Connecting to: date.jsontest.com:80
[hostByName] request IP for: date.jsontest.com
[hostByName] Host: date.jsontest.com lookup error: -5!
Connection failed
```

horendus, 2017-05-11:
Im having a similar issue, a bit different but maybe related. 

I find that sometimes my esp8266 webserver does not respond to GET requests for long periods of time, although the server and MCU is still running fine. 

I either keep trying and eventually it works or if I reboot my router it works really well for a while, and then starts randomly not responding again.

vlast3k, 2017-05-12:
i am also having similar issues, unfortunately i am unable to reproduce it
with my router, and it apparently happens only with some devices and some
routers, as i sporadically get some complaints from some people, where the
device stops sending data after few days of operation and then a hard
reboot fixes it.

did you try whether the git version behaves better in your case ?

WanaGo, 2017-06-11:
Same issue here.

Each time WiFi.hostByName() gets called after the initial time, I get the following:

[hostByName] request IP for: pool.ntp.org
[hostByName] Host: pool.ntp.org lookup error: -5!
pool.ntp.org: 0.0.0.0

But the first attempt I make using the same code, works correctly.

Transmit NTP Request
[hostByName] request IP for: pool.ntp.org
[hostByName] Host: pool.ntp.org IP: 202.6.116.123
pool.ntp.org: 202.6.116.123
Receive NTP Response

There is a bug in here somewhere. I don't know where though. Its a bit over my head.

WiFi.hostByName() seems to work once. So as a work around I tried to do this in setup(), and save the IP address of the NTP server I have selected by name initially from a pool address, and then just use that IP address for each update after that. 
However that didn't work. Each time it tried to use that IP, it doesn't work correctly. So I dont know if this is a hostByName() issue itself, it seems a little deeper, as specifying the IP to use doesn't work either.

me21, 2017-08-23:
I've got the same error here with lookup error -5 and it seems that esp_yield does not pause the execution of the function. I've added debug output before esp_yield, after it, and in callback function.

This is what I get:
```
before yield
after yield
in callback
```

ianleeder, 2018-02-08:
Complete ESP8266 newbie, apologies if I'm way off track of have an obvious error here.

My experience is the reverse of WanaGo.  On the first call to `http.GET()` I get the following failure every time:
`[hostByName] ... lookup error: -5!`.
But subsequent tries work fine.  For me it is 100% reproducible (which makes me think I'm doing something wrong).

I've uploaded my code to a gist here:
https://gist.github.com/ianleeder/3ff84c1564e94a371e53f41d83fdcb9f

In my code I find that my first `http.GET()` fails every time, the second works (Google gives a 302 redirect from HTTP to HTTPS, but it's still a valid connection).  With Debug Level set to `WIFI`, I get:
```
[hostByName] request IP for: www.google.com
[hostByName] Host: www.google.com lookup error: -5!
[Get] [5003ms] First try code -1, error message: connection refused
[hostByName] request IP for: www.google.com
[hostByName] Host: www.google.com IP: 216.58.200.100
[Get] [7116ms] Second try code 302, error message:
```

Weirdly though I tested doing a hostByName call directly, and it works fine.

arrowcircle, 2018-09-07:
Got same issues with my esp8266 boards. Got random dns lookout errors.
Running latest SDK via PlatformIO.
Also, reconnecting to wifi solves the problem for a few requests.
Any ideas how to solve this error?

arrowcircle, 2018-09-12:
Looks like changing lwIP flag to PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH solves the problem.

jholweg, 2019-03-28:
For me, the problem after a WiFi reconnect only happens with CNAME records. A records resolve fine. Any idea?