esp8266/Arduino #3152: ESP8266 (wemos d1 mini) OTA update WDT reset

radimsejk, 2017-04-20:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini
Core Version:      	        2.3.0

### Description

When I upload my code over the OTA, the ESP resets because of WDT after upload.

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB/ 1M(spiffs)
CPU Frequency:      80Mhz
Flash Mode:		??
Flash Frequency:	??
Upload Using:		OTA
Reset Method:		??

### Sketch

#include <ESP8266WiFi.h>
#include <ArduinoOTA.h>
//#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager

// OTA SETTING (PLEASE UPDATE THEESE VALUES)
const char* OTApassword         = "MYPASS";           // OTA auth password
String ID                       = "TEST_DELETE";      // ID of this module (put some name here)
int wifiNotConnected            = 0;                  // WiFi not connected counter

// OBJECTS INITIALISATION
WiFiClient espClient;

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.println();
  
  Serial.print("Connecting to WiFi");
  //Serial.println(ssid);*/

  ID += " ";
  ID += String(ESP.getChipId());
  char c[50];
  ID.toCharArray(c, ID.length() + 1);

  WiFiManager wifiManager;
  wifiManager.setTimeout(180);
  wifiManager.setDebugOutput(false);

  if (!wifiManager.autoConnect(c, "MYWIFIPASS")) {
    Serial.println();
    Serial.println("WM: failed to connect and hit timeout");
    delay(3000);
    //reset and try again
    
    delay(5000);
    ESP.restart();
    delay(5000);
  }
  else {
    Serial.print("...CONNECTED");
  }

  // print debug data
  Serial.println();
  Serial.println("SUMARY:");
  Serial.println("WIFI CONNECTED");
  Serial.print("IP ADDRESS: ");
  Serial.println(WiFi.localIP());

  // setup OTA
  char hostname[50];
  ID.toCharArray(hostname, ID.length() + 1);
  ArduinoOTA.setHostname(hostname);
  ArduinoOTA.setPassword(OTApassword);
  ArduinoOTA.onStart([]() {
    Serial.println("Starting OTA");
    

  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("OTA recievenig progress: %u%%\r", (progress / (total / 100)));
    for (int i = 0; i < (progress / (total / 100)); i++) {
      Serial.print("*");
    }
    for (int i = 0; i < 100 - (progress / (total / 100)); i++) {
      Serial.print("_");
    }
    
    Serial.println();
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();

  
  Serial.println(" READY TO ROCK!");
  Serial.println("---------------------------------");
  Serial.println();
}

void loop() {
  // get actual "time"
  unsigned long now = millis();

  // check WiFi connection
  if (WiFi.status() != WL_CONNECTED) {
    wifiNotConnected++;
    Serial.println("WiFi DOWN");
  
    delay(500);
    if (wifiNotConnected > 10) {
      Serial.println("Restarting");
      ESP.restart();
    }
  }

  // Over The Air updates loop
  ArduinoOTA.handle();
}

### Debug Messages

```
OTA recievenig progress: 100%
****************************************************************************************************

End

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

devyte, 2017-09-07:
@radimsejk did you ever get this work?

owarek, 2017-09-07:
Sadly not.

devyte, 2017-10-24:
@radimsejk after flashing over serial you have to physically reset the esp before attempting OTA, or you get your symptoms. Did you do that?

gaborpeltzer, 2017-11-25:
I have similar problem. My solution is call the setup funcion.

radimsejk, 2017-11-28:
When do you call the setup? In .onEnd([]() ? I will try it at this weekend.

gaborpeltzer, 2017-11-30:
Instead of ESP.restart(). Check this http://bit.ly/2j5veGx

devyte, 2019-11-06:
This thread is old, a lot has been fixed since it was opened.
Closing.
If you still encounter the problem, please open a new issue and follow the template instructions.

closed 2019-11-06 by devyte