esp8266/Arduino #3154: not working

jayswag2017, 2017-04-21:
i bought the latest ESP8266 im new to coding and arduino i did all the steps right but when i try to upload it to the board, it says:
error: espcomm_open_mem failed.

the error code is this:
Arduino: 1.8.3 Hourly Build 2017/04/14 10:33 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"
Sketch uses 542206 bytes (51%) of program storage space. Maximum is 1044464 bytes.
Global variables use 53504 bytes (65%) of dynamic memory, leaving 28416 bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM1
error: espcomm_open failed
error: Failed to open COM1

error: espcomm_upload_mem failed

And when i plug it in it wont download the drivers Please need Help

JamesGKent, 2017-04-21:
well if you aren't able to install the driver, that would explain why you are unable to program it. you need to install the drivers so that your operating system recognizes the ESP as a com port, and then tell the arduino ide to use that com port.

what happens when you try and install the drivers?
have you looked in the device manager to try and troubleshoot the device?

jayswag2017, 2017-04-21:
when i try to download it it says my driver is up to date

JamesGKent, 2017-04-21:
if you open up the device manager can you see an unknown device?

jayswag2017, 2017-04-21:
yes

JamesGKent, 2017-04-21:
can you confirm exactly what ESP model you are using? the ESP device comes in quite a few formats depending on who made the board. this will affect what usb to UART bridge chip is fitted. however it is possible that this bridge chip is broken, or its possible that your USB ports cannot deliver the required current, meaning the ESP tries to power up, draws too much current and the board will "brown out"

jayswag2017, 2017-04-21:
The 8266

jayswag2017, 2017-04-21:
As soon as i stick it in it says unknow device on all 3 of my computers

JamesGKent, 2017-04-21:
no, all esp8266 devices use the same/very similar chipset, but it could be a wemos board, a nodemcu or one of many others. have you tried connecting it through a powered usb hub? (would suggest one with at least a 1A but a 2A supply would be better)

jayswag2017, 2017-04-21:
I have the node MCU and I have tried it in a powered USB hub I changed the cord I was using and it worked in one computer which is alot newer than the others but it dosnt work on the other

jayswag2017, 2017-04-21:
i got it to work thank you so much for your help

closed 2017-04-29 by igrr