esp8266/Arduino #3156: Exception errors even on basic sketch

KyleHunter, 2017-04-21:
### Basic Infos

#### Hardware
Hardware:			ESP-12

### Description

Having tons of issues with my first attempts at using this board. I have a lot of experience using arduino's and various microcontrollers, but my current project requires WiFi connectivity. I ordered the esp-12 and have had trouble getting it to be stable at all. I'm using the arduino IDE.

No matter what sketch I use, even if it just a loop printing to serial, it is unbelievably unstable. It will print fine for a while, then non stop exception errors and stack traces. The exception numbers vary each time, usually 0/29/8. It's just super intermittent. It will work fine for 20 minutes and then just spam errors until I reset and re upload my sketch. I notice that it occurs almost every time I reset the board. I started using it just as it came without flashing any new firmware. I thought that it could be the issue, but then I flashed the lasted nodemcu, and there was no change.

I assume it is something with my power setup, since it seems like they are very sensitive, but I think I am doing it right. I'm using this schematic for the board:
esp8266.github.io/Arduino/versions/2.0.0/doc/ESP_improved_stability.png

A CP2102 Serial cable is used to interface with the board

Voltage is regulated with a LD1117V33 voltage regulator, from a 2AMP 5V wall power supply.

I connect the serial cable's 3.3v power to my common power rail, because it doesn't upload without that (I feel like this may be my issue).

I'm really bad at drawing schematics, so here's a picture of my voltage regulator board and the esp board. To upload I bring GPIO0 to low and have reset low and as it starts to upload bring it high. Thanks a lot for any help.

[![Board wiring][1]][1]

[![Voltage Regulator][2]][2]

  [1]: https://i.stack.imgur.com/RYbTL.jpg
  [2]: https://i.stack.imgur.com/zYE1I.jpg

### Sketch

```cpp

#include <Arduino.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Foo");
}

lrmoreno007, 2017-04-21:
I had many problems like the ones you are having and in the end I refused to use generic modules.

The problem was the interferences. I tried everything: several power supplies, power it with batteries, capacitors of all kinds and values, an insulated box ....

In the end, I bought a NodeMCU and finished all the problems. I believe that the two capacitors (100uF and 100nF) of the NodeMCU are the ones that create the magic, are located very close to Vcc (as Espressif recommends) and to be SMC type receive less interference.
[image]

Later and convinced that it was for this, I tried again with the generic module and breadboard, but the problems persist.

My advice is that if you do not want to break your head, buy a NodeMCU even if it is Chinese (or any other module already integrated).

Greetings.

KyleHunter, 2017-04-21:
Yeah but I plan to integrate this module in a production pcb. It's a tough option to have to a prefab module in a production part.. It's shocking to me that nodemcu is the only one who can manage to fix this since it seems like a common issue

lrmoreno007, 2017-04-21:
Do not! NodeMCU is not the only one, any other non-generic module should be able to work well.

And they all carry a Generic ESP8266, but not the breadboard and threads. In other words, really the problem is the breadboard and threads.

If you are going to make a PCB, keep in mind what I tell you about the distance between capacitors and Vcc, and also try to leave the Reset track as short as possible.

If you search on Google "esp8266 interference decoupling capacitor" you will find lots of literature about it. But they are always problems with generic modules.

Greetings.

KyleHunter, 2017-04-21:
So you're saying that even in a production part, I'm best to get a generic nodemcu rather than attempt to implement the esp 12 in my board? I guess that's doable

lrmoreno007, 2017-04-21:
Sorry if I do not explain myself very well, but I'm Spanish and I have to use the Google translator xD.

Integrating a NodeMCU or any other can be a good solution.

But really what I want to tell you is that if you choose to make your own PCB with a generic, you have to study very well the position of all the elements taking into account the two points that I comment to avoid interference:
- Short distance between Vcc and capacitors.
- Reset track as short as possible.

NodeMCU and many others are Open Hardware, so one option is to study how they are made and adapt it to your needs.

Greetings.

KyleHunter, 2017-04-21:
Perfect, thanks for your advice. Makes sense especially since my module is on a large breadboard with long power runs only using a 0.1 and 10 decoupling caps. I will transfer it to a perf board and see what happens. Thanks again.

jjssoftware, 2017-04-30:
ESP12F generic modules are pretty much all I use and they can be 100% completely stable. I stress "can be stable" because when you build a circuit with one these modules integrated, it's the rest of the circuit around the ESP12F that has direct impact on whether your circuit is solid or flakey rubbish. 

IMO the most important thing when building a circuit that includes an ESP12F generic module, or indeed any of the ESP8266 modules that are not pre-packages into something like a nodeMCU board, is the power supply part of the circuit and the actual power supply itself. 

Take a look at this image as an example of something that I knocked up on stripboard: https://jjssoftware.github.io/assets/img/widget-on-stripboard.jpg

This is an an example of an ESP12F mounted on one of the widely available white breakout boards which has then been included in a slightly larger circuit which includes a few capacitors, resistors, a 433mhz transmitter and a DHT22 temp/humidity sensor.

The 2 large caps that you can see either side of the regulator are 100uF electrolytic caps and there's also a 0.1uF ceramic cap (that you can't see in this image because it's under the white break out board) connected directly between the ESP12F vcc and ground pins. I've heard some people use even larger caps than I've used here but I guess what you need depends on the layout of included components and perhaps other peripheral components on board that make demands on the power supply. The capacitors included work to smooth the spikes and filter the supply to keep the ESP happy. The 5v->3.3v regulator I'm using here is an LF33ABV LDO power regulator but any other regulator that can deliver 500mA+ is going to be fine - if cautious just go for a regulator that can deliver 1amp. I then have this powered via a 240v -> 5v adaptor that is capable of delivering 2amps. 

I can promise you that this stripboard circuit is 100% stable as you see it built here and this is powered up and operating 24/7. I have it monitoring ambient conditions at home, it controls my heating system and Mrs jjssoftware would not be happy at all if our heating didn't work :)

ESP8266 devices are power hungry little beasts at times and are very spikey in their demands on the power supply when WiFi is starting up and is actually in use. If your power supply is not up to the task, you can easily experience a brown out and a subsequent ESP board reset.

KyleHunter, 2017-04-30:
@jjssoftware
Seems spot on. I redid the circuit with smaller wire runs, and added a 100uF cap. Have ran it for the last week or so and I've yet to have it reset a single time.

Do you call ESP.restart() every so often just to be careful or no?

jjssoftware, 2017-04-30:
Great, sounds like a problem solved. 

I don't regularly call ESP.restart() on a fixed interval in my stuff. Basically the only times I call this are:

1. When some runtime configuration parameters are changed (by me) and saved to flash which then requires a restart to get everything configured correctly on boot. This is not very often at all - pretty much never once I have a device setup in the way I want it.
2. When the NTP client part of my stuff is unable to retrieve the current time after a number of retries. I do experience this occasionally, perhaps once a week.

Generally speaking heap memory usage should be fairly flat / stable / hovering around the same value at runtime and that will be the case so long as there's no memory leak. If this is the case and the power supply is solid, theoretically an ESP device should be able to run indefinitely without needing to be restarted.

KyleHunter, 2017-05-02:
Solved

closed 2017-05-02 by KyleHunter