esp8266/Arduino #3159: What is this ???

GR-noob, 2017-04-21:
Hello. I have this problem and i dont know what is it. I have all ready put the program on it but when i go to 192.168.4.1 at the APs page is refresing the page all the time and cant scan and i tried diferent versions to uploading it again but now this is showing and cant upload anymore:

Arduino:1.6.7 (Windows 8.1), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

     results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                             ^

sketch\APScan.cpp:14:61: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\User\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\User\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

> 

What i can do to fix it ???
(sorry for my english google translate is dum some times)

neu-rah, 2017-04-21:
> note: candidate expects 1 argument, 2 provided

the functions scanNetworks just expects a bool, you are passing 2 parameters to it

GR-noob, 2017-04-21:
Thanks for your time. It work and it pass this and is this:

Arduino: 1.6.7 (Windows 8.1), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

sketch\Attack.cpp: In member function 'bool Attack::send()':

Attack.cpp:102: error: 'wifi_send_pkt_freedom' was not declared in this scope

   if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){

                                                 ^

exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

neu-rah, 2017-04-21:
this one might be because you are missing and include on that file, but i'm not sure

GR-noob, 2017-04-22:
I have not change anything from the script or delete. The only thing i change is scanNetworks only.

neu-rah, 2017-04-22:
and you can put it back if you think its the problem, however you must be missing some #include ... on the top of your file, that's my bet

jaromanda, 2017-04-22:
Have you installed version 2.0.0 (no later) of this package, and edited user_interface.h as described in the ESP8266 deauther instructions?

neu-rah, 2017-04-22:
not at all, just interpreting c++ error messages

GR-noob, 2017-04-22:
Hello again. Thanks again everyone for your help. At the end i rage quit and i delete it all and reinstall it again. The version it was 2.3.0 because updated. Now it is working. It scan and find my network, attack is working. For what is the clients page ??? Everytime i scan it disconnect and when it go to reconnect my network is gone and nothing on clients page.

closed 2017-04-29 by igrr

divyank17, 2017-07-15:
exit status 1
'wifi_send_pkt_freedom' cannot be used as a function