esp8266/Arduino #3160: How to send a string to a external server(web service) with esp8266

024N, 2017-04-22:
Hi,
I want to send sensor values to a web service which is inside an external server. Arduino and Esp8266 will work with adaptor. It will connect to a computer(it has web server) with connectify. For example,
after I enter http://192.123.123.1:8080/Sample/res/door0 to browser, I can see door0 on server but i want to send it(http://192.123.123.1:8080/Sample/res/door0 or door1) with arduino uno if sensor value changes.
How can I do it with arduino uno? 
Thank you.

scrapes, 2017-04-24:
I would reprogramm the ESP8266 to Work on specific commands.
I would connect the Serial lines to the Arduino and Listen to Serial Data.
Arduino -(Serial CMD:"VALUE HI")--> ESP8266 

See the WebClient Example for HTTP connection.

Serial Listening:

```
const byte numChars = 128;
char receivedChars[numChars]; // an array to store the received data

bool newData = false;

void recvWithEndMarker() {
 static byte ndx = 0;
 char endMarker = '\n';
 char rc;
 
 // if (Serial.available() > 0) {
 while (Serial.available() > 0 && newData == false) {
 rc = Serial.read();

 if (rc != endMarker) {
 receivedChars[ndx] = rc;
 ndx++;
 if (ndx >= numChars) {
 ndx = numChars - 1;
 }
 }
 else {
 receivedChars[ndx] = '\0'; // terminate the string
 ndx = 0;
 newData = true;
 }
 }
}

bool showNewData() {
 if (newData == true) {
 Serial.println("NewData");
 Serial.println(receivedChars);
 newData = false;
 return true;
 }
 return false;
}

void loop(void)
{
  while(!showNewData()){
  recvWithEndMarker(); 
  delay(1000);
  }
  //Here a Stringcompare to your command, then the HTTP request.
}
```

Another way would be I2C but implementing a decent solution would be more complicated. 

Also reconsider if you need the Arduino.

Les-A, 2017-04-25:
I am doing something like that with only an ESP8266 and an MCP3008 ADC device that connects via SPI.  If you're only interested in digital (on/off) signals then there are equivalents to the MCP3008.  I'm replacing an Arduino Uno.  There's an example sketch on the Arduino site to repeatedly send data to a server and also for using SPI.

devyte, 2017-09-07:
This is not the right place to ask for assistance for your particular project. This is an issue tracker meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for assistance.
Closing.

closed 2017-09-07 by devyte