esp8266/Arduino #3161: wificlientsecure or httpclient error stack trace shows file paths with Igor's (@igrr) personal directories

electronicsguy, 2017-04-22:
Hello,
       I am using esp01 and esp01E. Whenever I have an error while using libraries like _wificlientsecure_ or _httpclient_, this is what I see in the resultant stack trace:

```
0x40223c47: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x402248a9: bi_divide at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 430
0x4010020c: _umm_free at C:\Users\Sujay Phadke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4020398c: ax_port_realloc at C:\Users\Sujay Phadke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 605
```
I understand why it points to some of my files. But why are there paths leading to Igor's personal directories? (@igrr). Are there any hard-coded paths in libraries like **axtls**, etc?

Functions _more_comps()_ and _bi_divide()_ in **bigint.c** seems particularly problematic. Could someone shed any light on this?

Complete stack traces located here:
[stack-trace-1](https://pastebin.com/Xxhhiha8)
[stack-trace-2](https://pastebin.com/ZPJpHBMM)
[stack-trace-3](https://pastebin.com/2x5xuSLY)

davisonja, 2017-04-25:
Typically such foreign paths come from libraries with debug information
compiled in which reference the files that were used to build the library.

igrr, 2017-04-29:
Debug info is now stripped from path prefixes, as of the latest master.

closed 2017-04-29 by igrr