esp8266/Arduino #3162: Use of DNSServer causes WiFi-Mode set to WIFI_AP_STA

computerlyrik, 2017-04-22:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-01
Core Version:      	2.3.0

### Description

Usage of DNSServer causes WiFi-Mode set to WIFI_AP_STA

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB/128k
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL (both)
Reset Method:		manual re-plug to get into flash mode

### Sketch

```cpp

#include <Arduino.h>
#include <DNSServer.h>
IPAddress apIP(192, 168, 4, 1);
const byte DNS_PORT = 53;
DNSServer dnsServer;

void setup() {
//IF Configuration exists
    WiFi.begin(ssid.c_str(), password.c_str());
//ELSE
   WiFi.softAP(s,p);
   dnsServer.start(DNS_PORT, "*", WiFi.softAPIP());
}

void loop() {

}
```

### Debug Messages

Setup is a bit more complex, but a description of the basic flow is easy:
- As there is no configuration (e.g. held in EEPROM or SPIFFS) 
=> Run AP Mode and start DNS server 
=> This causes WiFi Mode to get into mode 3
- Setup some Wifi-Credentials via Web-Request or else and reboot chip
=> ESP will connect to the configuration provided AND start it's Wifi-AP (Because AP_STA fits Wifi.begin());

### Workaround
Add ```WiFi.setMode(WIFI_STA);``` before using the credentials inside ```WiFi.begin(...)```.

You may add  ```WiFi.setMode(WIFI_AP);``` before starting the softAP / after running ```dnsServer.start(...)``` to have AP only mode - but this does not bother me at the moment.

computerlyrik, 2017-04-22:
PS: Without DNS-Server the Mode is switching between 1 and 2 because of ```WiFi.softAP``` and ```WiFi.begin``` are triggering (somehow?) a state change, if the current state does not fit for the intended connection.

igrr, 2017-05-09:
`WiFi.begin` will only enable `STA` if it isn't enabled yet. It will not disable AP. If you do need to disable AP, set the mode to `STA` using `WiFi.setMode`.

If it was the other way around, it wouldn't be possible to keep AP running when `WiFi.begin` is called. The way it is implemented now, both use cases are possible.

closed 2017-05-09 by igrr

computerlyrik, 2017-05-16:
So in which constellation should `WiFi.softAP` and `dnsServer.start` called on boot-up get the chip into WiFi-Mode 3?
I think it shouldn't. I am now writing a complete test-sketch to verify the issue and test it on a WeMos D1, on Success on an ESP-01.

As described it the issue post 

> Usage of DNSServer causes WiFi-Mode set to WIFI_AP_STA

The DNSServer seems to be the issue here.

I will post results here. 

Thank you

igrr, 2017-05-17:
For example:

1. After reset, mode is STA (loaded from flash)
2. WIFI.softAP is called. Mode is now AP_STA.
...

3. Next reset. Mode is AP_STA (loaded from flash)

If you add WiFi.mode(WIFI_AP) before the call to WiFi.softAP then you should get just the AP, without a STA.

computerlyrik, 2017-05-23:
I nailed it down in a small sketch - see https://github.com/computerlyrik/esp-wifi-mode-test

It works as expected - i do think it was the missing disabling of STA that caused the issue beforehand. STA will *still be enabled* after calling `WiFi.disconnect` so calling `WiFi.enableSTA(false)` is crucial.

Using `Wifi.mode(WIFI_AP)` is also possible - but in the case you want to have the client code separated from the AP code, disabling STA would be more lean.

DNSserver seems stable as everything else - at least with the latest updates.

Thank you very much for your help!