esp8266/Arduino #3163: Exception (3) Error Arduino (1.8.2) Core 2.3.0

hbwh19, 2017-04-22:
#### Hardware
Hardware:			ESP-12 (WeMos D1 mini clone)
Core Version:      	2.3.0

### Description

The Problem only happens then I try and use the ESP8266's WiFi.
It says Exception (3)... (Rest Bellow) in serial monitor.

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?

### Sketch

```
#include "ESP8266WiFi.h"
#include <Arduino.h>
#include <U8x8lib.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); 

void setup() {
  Serial.begin(115200);

  u8x8.begin();
  u8x8.setPowerSave(0);
  u8x8.clear();

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {;
  u8x8.setFont(u8x8_font_chroma48medium8_r);

  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  u8x8.clear();
  u8x8.setCursor(0,0);
  if (n == 0)
  {
    Serial.println("no networks found");
    u8x8.print("no WiFi");
  }
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    u8x8.print(n);
    u8x8.print(" WiFi P 1");
    int p = 1;
    int j = 1;
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");

      String ID = WiFi.SSID(i);
      ID.remove(15);

      u8x8.setCursor(0,j);
      u8x8.print(ID);
      u8x8.refreshDisplay();    // for SSD1606/7
      if ((j == 6) and (i > 6))
      {
        j = 0;
        delay(1000);
        p++;
        u8x8.clear();
        u8x8.setCursor(0,0);
        u8x8.print(n);
        u8x8.print(" WiFi P ");
        u8x8.print(p);
      }
      delay(50);
      j++;
    }
  }

  // Wait a bit before scanning again
  delay(5000);
  Serial.println("");
}
```

### Debug Messages

```
Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006f841 depc=0x00000000

ctx: cont 
sp: 3ffef330 end: 3ffef680 offset: 01a0

>>>stack>>>
3ffef4d0:  feefeffe feefeffe feefeffe feefeffe  
3ffef4e0:  feefeffe feefeffe feefeffe 3ffef5e0  
3ffef4f0:  00000484 00000484 000003fd 401004f4  
3ffef500:  00000000 00001000 000003fd 40106ff8  
3ffef510:  40004b31 3ffef540 0000001c 4021b549  
3ffef520:  40105666 4021b631 3fff0414 000003ff  
3ffef530:  000003fd 3ffef5e0 3fff0414 000003fd  
3ffef540:  ffffff00 55aa55aa 00000009 0000001c  
3ffef550:  0000001c 00000003 00000003 000003ff  
3ffef560:  4021ba20 3fff0414 3fff0414 000000ff  
3ffef570:  00000001 3ffef600 4021bba7 00000008  
3ffef580:  3fff0414 000000ff 3ffef5e0 00000000  
3ffef590:  3fff04d4 3ffef641 00000001 4021bc34  
3ffef5a0:  3ffef5e0 3fff0414 3fffdad0 3ffee650  
3ffef5b0:  3ffef600 3fff674c 3fff0414 3fffdad0  
3ffef5c0:  4021bc70 3ffee4cc 00000000 4020121a  
3ffef5d0:  40202de8 00000000 3ffee488 402024fc  
3ffef5e0:  00000000 00000017 3ffe8560 40201306  
3ffef5f0:  feefefaf 00000064 3ffee488 3ffee650  
3ffef600:  3fffda00 0000000f 3ffee488 402013a3  
3ffef610:  3fffdad0 00000008 3ffee488 402025c4  
3ffef620:  3ffef640 00080001 3ffee488 3ffee650  
3ffef630:  4021b296 00000001 3ffee624 3fffdad0  
3ffef640:  40202c43 00000000 3ffee488 40202575  
3ffef650:  3fffdad0 3ffee624 3ffee4cc 402029b2  
3ffef660:  feefeffe 00000000 3ffee648 40203c1c  
3ffef670:  feefeffe feefeffe 3ffee660 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

jjssoftware, 2017-04-30:
there's not much to go on here - perhaps it's a power supply problem.

igrr, 2017-05-09:
@hbwh19  Please use ESP Exception decoder to decode exception output.

earlephilhower, 2019-01-16:
Closing as no feedback and obsolete.

closed 2019-01-16 by earlephilhower