esp8266/Arduino #3165: Blink doesn't work

afkiss, 2017-04-23:
Hello! I just bought an ESP8266-01 WiFi moudle, and I uploaded your Blink sketch through Arduino IDE, but the LED’s not blinking at all. Please help me to solve my problem. Thanks.

Hardware:			ESP-01
Core Version:      	        ?        ?

Problem description
 
The ESP’s built-in blue LED lights up dimly, but nothing else happens. 

Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K(64K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

Sketch

```cpp
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

### Debug Messages

```

Sketch uses 222 201 bytes (51%) of program storage space. Maximum is 434 160 bytes.
Global variables use 31 576 bytes (38%) of dynamic memory, leaving 50 344 bytes for local variables. Maximum is 81 920 bytes.
Uploading 226352 bytes from C:\Users\Pc\AppData\Local\Temp\build8424718093711560218.tmp/Blink.cpp.bin to flash at 0x00000000
................................................................................ [ 36% ]
................................................................................ [ 72% ]
..............................................................                   [ 100% ]

```

jaromanda, 2017-04-26:
I've read that new variants of ESP-01 have the LED on GPIO2 - rather than GPIO1/Tx (LED_BUILTIN is defined as 1 in the generic variant of ESP8266)

Symsc, 2017-06-15:
I have the same Problem. Even switching GPIO1 to 2 didn`t work

devyte, 2017-09-06:
@igrr how are these types of issues usually handled? Should a new board be added?

igrr, 2017-09-06:
Not sure about this one. Maybe we can add another option, "Builtin LED pin", to the Generic board? It would be a bit odd to have both "Generic" and "Generic, LED on GPIO2" boards.

d-a-v, 2018-01-08:
Can you try the LED menu now available in the latest git ?

mkucukiba, 2018-05-01:
If used in your scetch; --> pinMode (LED_BUILTIN, OUTPUT);
This line will not work: --> serial.print ("xxxx").
You should choose one.

GeVanCo, 2018-05-21:
For what it's worth: I've currently connected a NodeMCU 0.9 (ESP-12 Module).  Using LED_BUILTIN as a pin number works fine to blink the **NodeMCU LED** (not the ESP-12 LED which is also on the board) on the NodeMCU 0.9 board.  However, when using the value 13 instead (as is told by many on the internet), it wasn't working.
But it's simply straightforward to know what's the value of LED_BUILTIN.  Just add this into your `setup()` routine and you'll know the answer:

`
void setup() {
  Serial.begin(115200);
  Serial.print("LED_BUILTIN = ");
  Serial.print(LED_BUILTIN, DEC);
}
`
In my case, LED_BUILTIN was not 13, but 16.  This is because the NodeMCU redefined the Arduino pins.

Note: for NodeMCU you can use the pins defined by NodeMCU too.  For instance, use D5 (NodeMCU pin) and it will be translated to the correct Arduino GPIO pin (in that case, gpio 14).

Here's the translation:

`static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;
`

d-a-v, 2018-09-03:
Led pin is now a choice in menu with generic boards. Closing.

closed 2018-09-03 by d-a-v

AndyUK1966, 2019-09-06:
#define MY_LED 1

setup()
{  
pinMode(MY_LED, OUTPUT);
digitalWrite(MY_LED, HIGH); //Off
)
loop()
{
  delay(500);
  digitalWrite(RELAY, LOW); // On
  delay(500);
  digitalWrite(RELAY, HIGH); //Off
}

wdmtech, 2020-05-07:
For anyone else that was looking for it like I was, you can configure the LED's GPIO here

[image]

VastSky, 2020-12-28:
Solution[](https://arduino.stackexchange.com/questions/29938/how-to-i-make-the-tx-and-rx-pins-on-an-esp-8266-01-into-gpio-pins)

It seems that GPIO1 (TX) and GPIO3 (RX) are by default TX & RX. 

`Add this to your code at the beginning of set void setup():

//********** CHANGE PIN FUNCTION  TO GPIO **********
//GPIO 1 (TX) swap the pin to a GPIO.
pinMode(1, FUNCTION_3); 
//GPIO 3 (RX) swap the pin to a GPIO.
pinMode(3, FUNCTION_3); 
//**************************************************
You will no longer be able to use the Serial Monitor as TX will now be a GPIO pin and not transmit Serial data. You can still Flash your device as when you boot the device in flash mode it converts GPIO1 and GPIO3 back to TX/RX. Once you reboot into regular running mode GPIO1 and GPIO3 will go back to being GPIO pins.

To change GPIO1 and GPIO3 back to being TX/RX for regular Serial Monitor use add this to your code at the beginning of set void setup():

//********** CHANGE PIN FUNCTION  TO TX/RX **********
//GPIO 1 (TX) swap the pin to a TX.
pinMode(1, FUNCTION_0); 
//GPIO 3 (RX) swap the pin to a RX.
pinMode(3, FUNCTION_0); 
//***************************************************`

SaitamaTechno, 2023-11-30:
My solution:
if you dont use Serial.begin(), then you can use gpio1 (blue_led) on esp01.