esp8266/Arduino #3166: web server problem

2aza, 2017-04-23:
### Basic Infos

#### Hardware
Hardware:			ESP8266-12MOD
Core Version:      	2.3.0
### Description

Problem description
Hi, i've try to write a little program with esp, it's work corretly but only on IE and Edge brosware, if i try the same program on chrome, firefox or another broswer it's very very slow to send the html data. Why?
Have nice day.
Filippo.
### Settings in IDE

Module:  			nodemcu 1.0
Flash Size: 		4MB
CPU Frequency:		160Mhz?
Flash Mode:			3m spiffs
Flash Frequency:	?
Upload Using:		 SERIAL
Reset Method:		nodemcu

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
//ip:192, 168, 4, 1
const char *ssid = "Incubatrice";
const char *password = "0123456789";
ESP8266WebServer server(80);
#include "DHT.h"
DHT dht(2, DHT22);//pin e tipo
#include "SSD1306.h"
SSD1306 display(0x3c, D3, D5);// D3 -> SDA D5 -> SCL
String temp;
String umid;
String giorni = "0";
String recordtempm;
String recordtempM;
String recordumidm;
String recordumidM;
unsigned long millisdht = 0;
unsigned long millisgiorni = 0;
float t, u, ruM, rum, rtm, rtM;
int memprimaaccenzione = 0;

void setup() {
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();

  server.on("/", []() {
    server.send_P(200, "text/html", index_html);
  });
  server.on("/temp", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt;'>" +  temp + "</p></html>");
  });
  server.on("/umid", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" +  umid + "</p></html>");
  });
  server.on("/giorni", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" +  giorni + "</p></html>");
  });
  server.on("/recordtempm", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" +  recordtempm + "</p></html>");
  });
  server.on("/recordtempM", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" +  recordtempM + "</p></html>");
  });
  server.on("/recordumidm", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" + recordumidm + "</p></html>");
  });
  server.on("/recordumidM", []() {
    server.send(200, "text/html", "<html><meta http-equiv='refresh' content='2'><p style='color:#1F497D;font-size:24pt'>" + recordumidM + "</p></html>");
  });

  server.begin();
  dht.begin();
  display.init();
}

void loop() {

 unsigned long millisattuali = millis();
 
  server.handleClient();
 

  /////////lettura temperatura e umidità + display///////////
  if (millisattuali - millisdht >= 2000) {
   
    millisdht = millisattuali;
    t = dht.readTemperature();
    temp = String(t, 1);
    u = dht.readHumidity();
    umid = String(u, 1);
    display.clear();
    display.setFont(Nimbus_Roman_No9_L_Regular_30);
    display.setTextAlignment(TEXT_ALIGN_LEFT);
    display.drawString(0, 0, "T");
    display.drawString(0, 35, "U");
    display.drawString(32, 0, temp);
    display.drawString(32, 35, umid);
    display.setTextAlignment(TEXT_ALIGN_RIGHT);
    display.drawString(128, 0, "c°");
    display.drawString(128, 35, "%");
    display.display();
    ///////////////////record temp. umid.////////////////////////
    if ( memprimaaccenzione == 0) {
      rtM = t;
      rtm = t;
      ruM = u;
      rum = u;
      memprimaaccenzione = 1;
    }
    if (t <= rtm) {
      recordtempm = String(t, 1);
    }
    if (t >= rtM) {
      recordtempM = String(t, 1);
    }
    if (u <= rum) {
      recordumidm = String(u, 1);
    }
    if (u >= ruM) {
      recordumidM = String(u, 1);
    }
  }
  ///////////////conta giorni incubazione//////////////////////
  if (millisattuali - millisgiorni >= 86400000) {
    millisgiorni = millisattuali;
    int gg = gg + 1;
    giorni = String(gg);
  }
}

```

horendus, 2017-05-08:
I can see your serving the index.html page from PROGMEM
How big is it?

I would try doing it from RAM to see if its faster as first test.

Also, perhaps the content-length header field is incorrectly generated. Does you index page show all of those other pages content within it somehow? If so the the original server.send_P generates the content-length header but does not take into account the extra length added by the other content sends you are making, which can cause some browsers to hang for a bit. 

You can try overriding it with setContentLength() but you need to calculate the correct length your self. 

Also try increasing #define HTTP_DOWNLOAD_UNIT_SIZE 1460 in ESP8266WebServer.h. Iv found increasing this can give lighting fast HTML serving although it does put the server out of spec with conventions, not a problem for general use.

igrr, 2017-05-09:
The sketch posted doesn't include `index_html` variable, could you please update the sketch?

closed 2017-12-28 by igrr