esp8266/Arduino #3167: delay causing (heap) memory loss

mcr-ksh, 2017-04-24:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 R2 & mini 
Core Version:      	?2.3.0?

### Description

Weird loss of heap memory when using delay(). 

I was trying to write to EEPROM and figured that after the write a heap memory structure was zeroed out. I suspected the problem to be the EEPROM but after intense digging I found that when commenting out the delay the memory isn't cleaned. 

### Settings in IDE

Module:  			?Wemos D1 R2 & mini?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?

### Sketch

```cpp

void write_SystemStatePartial(const void *addr, int len)
{
  int from = ((const byte*)(const void*)addr - (const byte*)(const void *)systemState);
  int to = from + len;
  if ( from < 0 ) {
    return;
  }

  // memory pointer
  const byte *ptr = (const byte*)(const void*)addr;
  
  for (int i = from; i < to; i++) {
    Debug("write_SystemStatePartial(): %d\t%02X\n", i, (byte)*(ptr));
    EEPROM.write(i, *(ptr++));
    //delay(100);
  }
  EEPROM.commit();

  //delay(500);
  Debug("bytes written = %d\n", to - from);
}

void caller()
{
    /* save current instructions */
    Debug("processResponse(): param1 = %s %x\n", kwr->todo->param1, kwr->todo->param1);  
    write_SystemStatePartial(systemState->cmd, strlen(systemState->cmd));
    Debug("processResponse(): param1 = %s %x\n", kwr->todo->param1, kwr->todo->param1);  
}

```

### Debug Messages
before commenting out delay.
```
processResponse(): param1 = 5000 3fff5947
write_SystemStatePartial(): writing from=115 to=119
write_SystemStatePartial(): 115	6E
write_SystemStatePartial(): 116	6F
write_SystemStatePartial(): 117	6F
write_SystemStatePartial(): 118	70
Mon Apr 24 15:36:04 2017

write_SystemStatePartial(): bytes written = 4
processResponse(): param1 =  3fff5947
```
the value of param1 is gone!
after commenting out delay.
```
processResponse(): param1 = 5000 3fff5947
write_SystemStatePartial(): writing from=115 to=119
write_SystemStatePartial(): 115	6E
write_SystemStatePartial(): 116	6F
write_SystemStatePartial(): 117	6F
write_SystemStatePartial(): 118	70
write_SystemStatePartial(): bytes written = 4
write_SystemStatePartial(): <--
processResponse(): param1 = 5000 3fff5947
```

igrr, 2017-05-09:
Thanks for the report. Could you please post a complete example which reproduces the issue, which someone else could compile? The code you have posted references some identifiers defined in other parts of your code. Thanks.

devyte, 2017-09-06:
@mcr-ksh will you be adding a complete example that reproduces the issue?

devyte, 2017-10-10:
Requested sketch, no reply in several months, closing.

closed 2017-10-10 by devyte