esp8266/Arduino #3168: ADC raw max value issue on ESP12E

davidegironi, 2017-04-24:
Hello, I've an issue similar to the already closed post here: https://github.com/esp8266/Arduino/issues/2672
I get ADC raw values lower than 1024 by analogRead(A0), when A0 is almost 1V.
Tested on NodeMCU v3 board. It has a voltage divider formed by 220k and 100k resistor attached to A0.
A0 is connected to 3.272 board output.
Using analogRead(A0) i get almost 978 value. I read 1.02V with a multimeter on A0 (which is a right voltage value according to the voltage divider), what's wrong is the ADC raw value? I expect it to be 1024, but it's lower.
Two different NodeMCU v3 board, and one NodeMCU Amica tested, same results.
ESP.getVcc() return 2.7V almost, on each of the 3 boards tested, it sounds strange.
Compiled and uploaded using Windows or Linux, doesn't make difference.
I use PlatformIO, but with Arduino IDE I get the same error.
WiFi on or off doesn't make difference.
Thanks for help.

Sample sketch below:
```
void setup() {
  Serial.begin(115200);
}

void loop() {
  int adcraw = analogRead(A0);
  Serial.println(adcraw);
  delay(1000);
}
```

lawrence-jeff, 2017-04-24:
Are you running the latest build out of the GIT or a released version? I believe the problem is still in the released 2.3.0 code.

davidegironi, 2017-04-24:
version=2.2.0 in platform.txt
core_esp8266_wiring_analog.c is the same of the actual GIT which is version=2.3.0 in platform.txt
Have to try with 2.3.0 downloaded from this repository?

lawrence-jeff, 2017-04-24:
No - 2.3.0 is too old. You need to follow the instructions in the section "Using git version" so you are running the latest code in the repository (the files that are days old not from June 2016)

davidegironi, 2017-04-25:
Thank you lawrence, I've test it with the latest clone of this repository, nothing changed.
I've solved it testing with a "plain" ESP-12E. On the plain ESP I get 1012 raw value for a 1.03V input and supply is 3.32V, the 1012 raw value is enough accurate to me for this ADC, for more precise values I should use an external ADC.
This issue can be closed to me as it is not a code problem, is much more an hardware (nodemcu board type) problem, had to investigate why.

closed 2017-04-28 by davidegironi