esp8266/Arduino #3171: Wifi.Begin password lowercase not working

SSShreyas, 2017-04-25:
### Basic Infos

#### Hardware
Hardware:			?ESP-8266?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			? ESP8266 Huzzah Feather Module?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

// Import required libraries
#include "ESP8266WiFi.h"

// WiFi parameters
const char* ssid = "Home1434-5G";
const char* password = "dynamicoctopus392";  // lowercase password 

void setup(void)
{ 
// Start Serial
Serial.begin(115200);
//WiFi.hostname("ESP8266");
// Connect to WiFi
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("WiFi connected");

// Print the IP address
Serial.println(WiFi.localIP());

}

void loop() {

}
```

### Debug Messages

Output on serial port 
rl l��| �l�|�l�b|����r�b�b��nn�lnn���bp��lrlrlp�n��l��bn�|l��b��nn� l��l`�nnl`nr���nb�lr��nb�l�l���l�l`��n�
..............

**If I connect to other ssid : Home_1434 with password : DYNAMIC392 "Uppercase password" then the same sketch works.**

devyte, 2017-04-25:
As I understand it, the ESP8266 is not 5G capable. Your ssid ending in -5G
hints that it's a 5GHz one.

SSShreyas, 2017-04-25:
Oh thanks a lot. I'll try it out.

closed 2017-05-09 by igrr