esp8266/Arduino #3172: How to Connect to open network

gregbridge, 2017-04-26:
Hey so super simple question but it is killing me. I'm trying to connect my board to an open wifi network (no password), what do I enter in the password field?

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

electronicsguy, 2017-04-26:
Have you tried with a blank string? You need to try and see what happens. In case of errors, post the output here. Also read this: [esp8266-blank-pass](https://forum.micropython.org/viewtopic.php?t=2971)

igrr, 2017-05-09:
`WiFi.begin(ssid)` or `WiFi.begin(ssid, NULL)` should work for open networks.

closed 2017-05-09 by igrr

adrionics, 2019-01-05:
WiFiMulti has an issue with this.

See https://github.com/esp8266/Arduino/issues/5593