esp8266/Arduino #3173: Can Monitor/Promiscuous Mode sniff devices not connected to an AP?

heinvlog, 2017-04-26:
I am currently running the NodeMCU ESP 8266 in promiscuous mode.
Setup below:
  wifi_set_opmode(STATION_MODE);            // Promiscuous works only with station mode
  wifi_set_channel(channel);
  wifi_promiscuous_enable(disable);
  wifi_set_promiscuous_rx_cb(promisc_cb);   // Set up promiscuous callback
  wifi_promiscuous_enable(enable);

It is not connected to an AP so effectively it is running in monitor mode.
All AP's and devices connected to the AP's is sniffed by the sniffer.
But, devices (with wifi switched on) not connected to an AP is not picked up. 
How can this be achieved ?

My design is based on the link below:
https://www.hackster.io/rayburne/esp8266-mini-sniff-f6b93a

devyte, 2017-09-07:
Your project sounds very interesting. I believe I've seen others do similar things on project sites in the past (don't remember details anymore). I hope you got it working.
However, this is an issue tracker, meant for tracking issues in the core, and not a general support forum. Please refer to esp8266.com or stackoverflow for assistance.
Closing.

closed 2017-09-07 by devyte

ezequiel9, 2019-03-21:
@heinvlog Have you find something to do it? Seems like promiscuous mode only can catch connected devices. Monitor mode should be able to detect all.