esp8266/Arduino #3177: NodeMcu (Amica)    wdtwdt reset

Ayushkhandelwal15, 2017-04-28:
I'm fetching data from my college website and displaying onto the LCD(20*4).
my code fetch the data first  time but after that, it shows
wdt reset load 0x4010f000, len 1384, room 16  tail 8 chksum 0x2d csum 0x2d v60000318 ~ld
#### Hardware
Hardware:		NodeMcu(Amica)

*
Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:	80Mhz

### Sketch
#include <ESP8266WiFi.h>
#include <Wire.h>  // This library is already built in to the Arduino IDE
#include <LiquidCrystal_I2C.h> //This library you can add via Include Library > Manage Library >  

 
LiquidCrystal_I2C lcd(0x27, 20, 4);

const char* ssid = "AAYUSH"; // put your router name
const char* password = "987654321";// put your password 
 
const char* host = "api.thingspeak.com";
8
 
void setup() {
  Serial.begin(115200);
  delay(100);
  lcd.init();   // initializing the LCD
  lcd.backlight(); // Enable or Turn On the backlight 
  delay(200);
   lcd.noBacklight();
   delay(200); 
    lcd.backlight(); // Enable or Turn On the backlight 
  delay(200);
   lcd.noBacklight();
   delay(200); 
    lcd.backlight(); // Enable or Turn On the backlight 
  delay(200);
 //  lcd.noBacklight();
   delay(200); 
   lcd.setCursor(0,1);
  lcd.print("Latest News");
  lcd.setCursor(0,2);
  lcd.print("Poornima university");
 
 // We start by connecting to a WiFi network
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
  
void loop() {
  
  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
   
  }
  
  // We now create a URI for the request
 // /apps/thinghttp/send_request?api_key=DCCV9I2Y8K39BJ7N
 
  String url = "/apps/thinghttp/send_request?api_key=K6B2492YPS3KC3GY";
  Serial.print("Requesting URL: ");
  Serial.println(url);
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(2000);
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available())
  {
  String line = client.readStringUntil('\r');   
  Serial.print(line);
  line.trim();
  int indexfirstcloseingbracket = line.indexOf(">");
  //Serial.println(indexfirstcloseingbracket);
  int secondcloseingbracket = indexfirstcloseingbracket + 1;
  int indexsecondcloseingbracket = line.indexOf(">",secondcloseingbracket);
  Serial.println(indexsecondcloseingbracket); 
  line.remove(0,indexsecondcloseingbracket);
  int indexfirstopeningbracket = line.indexOf("<");
  line.remove(indexfirstopeningbracket);
  Serial.println(line);
  
  }
   
  Serial.println();
  Serial.println("closing connection");
  delay(3000);
  }

on the serial monitor, it is showing 
wdt reset load 0x4010f000, len 1384, room 16  tail 8 chksum 0x2d csum 0x2d v60000318 ~ld

igrr, 2017-05-09:
You could possibly add some debugging output to see where the issue happens: in the part of the code related to the LCD, or the part of the code related to HTTP connection.
The output after `wdt reset` is not very useful, it is printed on every startup and doesn't give much information about the issue.

devyte, 2017-09-06:
@Ayushkhandelwal15 any progress with this?

closed 2017-12-28 by igrr