esp8266/Arduino #3178: Unable to create a project for D1 Mini

Altair31, 2017-04-29:
### Basic Infos

#### Hardware
Hardware:		d1_mini
Core Version:      	N/A

### Description

I wanted to use your command line tool but it gives me an error each time I try to create my project.
I followed your "[How to](http://docs.platformio.org/en/stable/ide/eclipse.html)" to create my project.

So when I run:

platformio init --ide eclipse --board d1_mini

I get:
```
The next files/directories have been created in C:\Users\<user_name>\MyCloud\Personnel\Side projects\Domotique 2\controler
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
Error: [04/29/17 11:56:11] Processing d1_mini (platform: espressif8266, board: d1_mini, framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Collected 23 compatible libraries
Looking for dependencies...
Project does not have dependencies

UnicodeDecodeError: 'utf8' codec can't decode byte 0xe9 in position 10: invalid continuation byte:
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 162:
print "\n%s\n" % json.dumps(env.DumpIDEData())
File "c:\python27\lib\json\__init__.py", line 244:
return _default_encoder.encode(obj)
File "c:\python27\lib\json\encoder.py", line 207:
chunks = self.iterencode(o, _one_shot=True)
File "c:\python27\lib\json\encoder.py", line 270:
return _iterencode(o, 0)
 [ERROR] Took 1.11 seconds
```
or in another path without extended ascii character nor spaces:
```
The next files/directories have been created in C:\Temp\pioTest
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\init.py", line 114, in cli
    pg.generate()
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 116, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\python27\lib\site-packages\platformio\ide\projectgenerator.py", line 122, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "c:\python27\lib\site-packages\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "c:\python27\lib\site-packages\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "c:\python27\lib\site-packages\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 4, in <module>
  File "c:\python27\lib\site-packages\bottle.py", line 3347, in <lambda>
    self._escape = lambda x: escape_func(touni(x, enc))
  File "c:\python27\lib\site-packages\bottle.py", line 123, in touni
    return s.decode(enc, err) if isinstance(s, bytes) else unicode(s)
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xe9 in position 1306: invalid continuation byte

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
### Settings in IDE

No IDE yet :(

### Sketch

no sketch yet

### Debug Messages
See description

igrr, 2017-04-29:
Since you are reporting an issue with platform.io environment, would you mind opening the issue on that project's tracker instead? Thanks.

closed 2017-04-29 by igrr

Altair31, 2017-04-29:
Indeed !
Sorry