esp8266/Arduino #3179: Headers collection is empty

ClaudioHutte, 2017-04-29:
### Basic Infos

#### Hardware
Hardware:			ESP8266-12E
Core Version:      	2.3.0

### Description

Reading the headers collection ( i.e. calling server.headers() ) does not return any item (apart "Authorization"), despite the client sent several headers. For instance the example provided in SimpleAuthentification.ino
which read cookies does not work (server.header("Cookie"); always returns an empty string).

Am I missing something?

Also, I've looked into the three files ESP8266WebServer.cpp, ESP8266WebServer.h and Parsing.cpp, and I don't get how the header collection works. To my understanding the relevant parts are those listed below.
First, into the constructor(s) the variable _headerKeysCount is initialized to zero; then invoking the method begin() in turn it will call collectHeaders:
```cpp
void ESP8266WebServer::collectHeaders(const char* headerKeys[], const size_t headerKeysCount) {
  _headerKeysCount = headerKeysCount + 1;
  if (_currentHeaders)
     delete[]_currentHeaders;
  _currentHeaders = new RequestArgument[_headerKeysCount];
  _currentHeaders[0].key = AUTHORIZATION_HEADER;
  for (int i = 1; i < _headerKeysCount; i++){
    _currentHeaders[i].key = headerKeys[i-1];
  }
}
```
which initializes the collection with just one item and preset it to AUTHORIZATION_HEADER.

Then in Parsing.cpp:
```cpp
bool ESP8266WebServer::_parseRequest(WiFiClient& client) {
  // Read the first line of HTTP request
  String req = client.readStringUntil('\r');
  client.readStringUntil('\n');
  //reset header value
  for (int i = 0; i < _headerKeysCount; ++i) {
    _currentHeaders[i].value =String();
   }
```
(omissis)
```cpp
    String headerName;
    String headerValue;
    //parse headers
    while(1){
       /* (omissis) */
       headerName = req.substring(0, headerDiv);
      headerValue = req.substring(headerDiv + 2);
      _collectHeader(headerName.c_str(),headerValue.c_str());
      /* (omissis) */
      }
    }
```
thus in _collectHeader to my understanding it seems there is no way the new header will be collected since 1) the _headerKeysCount should still be 1, and 2) the _currentHeaders[0].key should be already set to AUTHORIZATION_HEADER. So I wonder how this function was meant to work.
```cpp
bool ESP8266WebServer::_collectHeader(const char* headerName, const char* headerValue) {
  for (int i = 0; i < _headerKeysCount; i++) {
    if (_currentHeaders[i].key==headerName) {
            _currentHeaders[i].value=headerValue;
            return true;
        }
  }
  return false;
}
```
For sure I'm missing some point, so I'm kindly asking for a little help to understand how this was intended to work and how to resolve the problem I met in retrieving the headers from the client. Thank you.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP12E)
Flash Size: 		4MB (3MB spiffs)
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?40Mhz?

### Sketch

Remark: this is just a snippet of the sketch, that I extracted from a much bigger program, in the attempt to focus on this specific issue more easily.

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <FS.h>

ESP8266WebServer server(80);

void senddefault() {
        server.sendHeader("Location","/testheaders");
	server.sendHeader("Cache-Control","no-cache");
	server.sendHeader("Set-Cookie","hello");
	server.send(302);
}

void testheaders() {
  /*  This will print out the following:
       num headers: 1
       header: Authorization = 
  */
  Serial.printf("num headers: %d\n",server.headers());
  for(int i=0;i<server.headers();i++) {
     Serial.printf("header: %s = %s\n", server.headerName(i).c_str(), server.header(i).c_str());
  }
  server.send(200, "text/plain", "test headers");
}

void setup() {
   Serial.begin(74880);

   /* here the code to inizialize wifi (omitted) */

   server.on("/testheaders", testheaders);
   server.on("/", senddefault);
   server.begin();
}

void loop() {
   server.handleClient();
}
```

jjssoftware, 2017-04-30:
collectHeaders isn't automatically called to gather the headers you're interesting in later using, you need to call this yourself e.g in setup. 

There's an example here: https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino#L124

ClaudioHutte, 2017-04-30:
Ouch I missed those lines in the setup. That makes sense about how the collection of headers works.
Thank you so much @jjssoftware .

closed 2017-04-30 by ClaudioHutte

WillingLearner, 2018-01-22:
Hello, I am new in programming ESP8266 and I am trying to understand how the codes work. I have a question about what server.collectHeaders method is used for ?