esp8266/Arduino #3181: Arduino NodeMCU POST not working

Raziel422, 2017-05-01:
I have a `NodeMCU 1.0` with `ESP-12E` on it and using Arduino IDE for writing code on it.  

I want to send a `POST` message to a remote `php` page on my webserer containing values from my module, but firstly I need to send something...  

I tried many different examples from different sources and code snippets, but none seem to work. The method it's being sent 
The `GET` method it's working, I can retrieve data from php pages, but I cannot send to them.

My code:  

    #include <ESP8266WiFi.h>
    #include <ESP8266HTTPClient.h>
    #include <WiFiClient.h>

    const char* ssid     = "SomeWireless";
    const char* password = "12345";

    String server = "www.example.net"; // http://www.example.com

    void setup() {
     Serial.begin(115200);
      delay(10);
     // We start by connecting to a WiFi network
      Serial.println();
      Serial.println();
      Serial.print("Connecting to ");
      Serial.println(ssid);

      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
  
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
      }

      Serial.println("");
      Serial.println("WiFi connected");  
      Serial.println("IP address: ");
      Serial.println(WiFi.localIP());
  
    }

    void loop() {
      jsonPOST();
      myPOST();
      myGET();
    }

    void myPOST() {
      HTTPClient http;
  
      http.begin("http://example.net/asd/recv.php");
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.POST("title=foo");
      http.writeToStream(&Serial);
      http.end();
  
      delay(1000);
    }

    void jsonPOST() {
      WiFiClient client;
  
      if (client.connect("example.net", 80)) {
          Serial.println("Connected to server");
      // Make the HTTP request
          int value = 2.5;  // an arbitrary value for testing
          String content = "{\"JSON_key\": " + String(value) + "}";
          client.println("POST /asd/recv.php HTTP/1.1");
          client.println("Host: example.net");
          client.println("Accept: */*");
          client.println("Content-Length: " + String(content.length()));
          client.println("Content-Type: application/json");
          client.println();
          client.println(content);
      }
      delay(1000);
    }

    void myGET() {
      if (WiFi.status() == WL_CONNECTED) { //Check     WiFi connection status
        HTTPClient http;  //Declare an object of class HTTPClient
        http.begin("http://www.exmaple.net/asd/btnCheck.php");  //Specify request destination
        int httpCode = http.GET();  //Send the request
 
        if (httpCode > 0) { //Check the returning code
          String payload = http.getString();   //Get the request response payload
          Serial.println(payload);                     //Print the response payload
        }
        http.end();   //Close connection
      }
      delay(1000); 
    }

My php:

    $test = $_POST['title'];
    echo $test;

    //JSON
    $json = file_get_contents('php://input');
    $obj = json_decode($json);
    echo $obj;

Any advice on this matter, it's worth it. I've been beating my head with it, for 2 days.  
Thanks!

jjssoftware, 2017-05-01:
I can't debug your code from here but I'd suggest going back to basics:

* Verify the web server being posted to does have the ability to accept and process a POST request
* Create a web page with a form element that can perform a POST and test the request to the web server that way
* If you control the web server code / web server configuration, make a change to make this arrangement easy to debug i.e. log requests to a file
* Set up a way to diagnose traffic on the network i.e. a proxy server or wireshark and compare the POST request sent from your test web page to the request sent from your ESP device

Raziel422, 2017-05-01:
Hi @jjssoftware 

- I have the ability to accept and process POST request. I can make this happen with jQuery and AJAX from a regular page and send the request to the specific page.
- Unfortunately, I don't control the web server, I have a reseller type of hosting.

Raziel422, 2017-05-01:
**ANSWER:**  

I manage to do the request properly. The problem was from my post content.  
The code that works for me with the `ESP8266HTTPClient.h` library is:

    void jPOST() {
      HTTPClient http;

      http.begin("http://example.net/asd/urlen.php");
      http.addHeader("Content-Type", "application/x-www-form-urlencoded");
      http.POST("{\"y\":6,\"x\":11}");
      http.writeToStream(&Serial);
      http.end();

      delay(2000);
    }

And for the php page:

    $req1 = file_get_contents("php://input");
    $req2 = json_decode($req1);

    $val= $req2 ->x;
    echo $val;
    //echo $req2 -> x;

WheresWally, 2017-05-12:
Hi, I found tthat when sending to a hosr with a shared IP you need a few headers so the webserver knows which virtual server to redisect to. And try GTTP/1.1 s well,

    client.println(" HTTP/1.1");
    client.println("Host: example.net:80");
    client.println("Connection: close");

devyte, 2017-09-06:
Given the last comment from May 1st, I consider this solved, so closing.

closed 2017-09-06 by devyte