esp8266/Arduino #3185: Only 5 TCP sockets

hcs-svn, 2017-05-01:
As far as I found out until now the default for the maximum number of concurrent TCP sockets ist 5.
Is there a way to increase it?

I see that the ESP32 SDK has an espconn_tcp_set_max_con method but I can not get it working.
Results in linker error: "undefined reference to espconn_tcp_set_max_con(unsigned char)"

Is it possible to use this method to increase the number of concurrent TCP sockets?

jjssoftware, 2017-05-01:
does this help: https://github.com/esp8266/Arduino/issues/570 (see `max_connection`)

hcs-svn, 2017-05-01:
Thanks, that pointed me to the right direction.

```
extern "C" {
  #include "espconn.h"
}
```
did the trick and
`espconn_tcp_set_max_con(8);`
allows now 8 concurrent TCP clients

closed 2017-05-01 by hcs-svn

fpalarmini, 2018-10-08:
This function "espconn_tcp_set_max_con(uint8 num)" did not work for me. Always returns -12 (invalid argument)... Does anyone have a suggestion? 
The version of the toolchain is 2.4.1.

d-a-v, 2018-10-08:
`espconn` is not tested with lwip2. Could you try with lwip1.4 in tools menu ?
Why don't you use arduino's `WiFiServer` / `WiFiClient` ?