esp8266/Arduino #3186: Watchdog reset on certain modules

tpillera, 2017-05-02:
### Basic Infos
Get WDT reset on certain modules with same software load.

#### Hardware
Hardware:			NodeMCU ESP-12 v0.9

### Description
I've had my code running fine on several units for a long time but on a newly purchased batch of v0.9 NodeMCU modules I get a watchdog reset as soon as I connect to the device over WiFi,

### Settings in IDE
Module:  			NodeMCU v0.9
Flash Size: 		4MB/1MB
CPU Frequency:	both 80 and 160 Mhz exhibit the problem

### Sketch

void setup() {
//Our ESP8266-12E is an AccessPoint.
WiFi.mode (WIFI_AP);
WiFi.softAP(ssid, password, channel);
}

void loop() {
server.handleClient();
}
```

### Debug Messages

Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld

Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld

ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room tail 8
chksum 0x2d
csum 0x2d

igrr, 2017-05-09:
Could you please mention which software version are you using? Stable 2.3.0, or latest git version?

If anyone can run this on NodeMCU 0.9 that would be appreciated, i don't have such boards anymore.

struthio, 2017-05-10:
I had same issue few days ago. 
I have home made board (with ESP12E) which I was using for some time without problems.

Few days ago I decided to replace ArduinoOTA in favor of updates through HTTP. After flushing it with new FW (flashing was done by UART) my device started to throw 'wdt reset' all the time.

I thought it was problem with my sketch, so I erased device with esptool and flashed code similar to Yours (just Wifi Connection), but device kept resetting.
Then I decided to disassemble my board (because probably some component was failing and causing power issue).

I ended up with device like https://github.com/esp8266/Arduino/blob/master/doc/ESP_to_serial.png and issue was not solved (I even replaced ESP12E with new one).

And out of the sudden Yesterday it worked! No resets. Even old codes work without resetting device.
I re-soldered most of elements back and device works as supposed to!

(Just to be clear. Before every test attempt I recompiled project from source and re-uploaded it to device).

dodedodo33, 2017-05-24:
Something you should add in the above sketch is probably a yield() call or any delay(milliseconds) call (yield() is a delay(0) call). This yields the execution to other processes running in the background, this is needed for the execution of the wifi stack and other functions. So if you're blocking execution for too long the wdt will trigger. The wdt (watchdog timer) is a timer that is reset during execution of some arduino framework code. If this timer is not reset in time the timer will trigger and the chip is reset. I've experienced similar issues to yours when I had a loop without a single delay or yield call.

devyte, 2017-09-08:
@tpillera is this still an issue? I'm not aware of any current problems with flashing to a nodemcu modules.

closed 2017-12-28 by igrr

tpillera, 2017-12-28:
Just an update...I've seen this now on several of the NodeMCU modules, more frequently on the 0.9 version but also on the 1.0's.  I've had them work fine for days then out of the blue start resetting for no apparent reason.  I believe it to be more of a quality/hardware issue with the part vs. some software problem but as a result I sidelined the project and will be exploring other options.  Possibly they have fixed this by now and the ESP8266 has a great feature-set but I am very hesitant to rely on this part.