esp8266/Arduino #3187: OTA Example fails

Thijxx, 2017-05-02:
It starts with the OTA protocol but not with sending any data.
This problem is the same in platformio and Arduino IDE.

```
Ready
IP address: 192.168.178.24
Start
Progress: 0%
Error[2]: Connect Failed
Error[4]: End Failed
```

### Basic Infos

#### Hardware
Hardware:		ESP-12

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		ck

### Sketch

```cpp
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Messages
Message read with serial connection while uploading using OTA.
```
Ready
IP address: 192.168.178.24
Start
Progress: 0%
Error[2]: Connect Failed
Error[4]: End Failed
```

mmaret, 2017-05-04:
The sketch you provide does not handle the OTA, it looks like the Blink example.
The BasicOTA sketch look like this https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino

Thijxx, 2017-05-04:
Yes :) I used that but this sketch above is the one that failed to load into the 'OTA enabled' device. The BasicOTA.ino is indeed the one I use.

mmaret, 2017-05-04:
Ho, sorry for the misunderstanding. :)
You could get more debug information by enabling the DEBUG_ESP_OTA flag.
I did not have my arduino gui in front of me, but IIRC, in the board menu, you have something like debug port to enable this. 
You could also run the script https://github.com/esp8266/Arduino/blob/master/tools/espota.py by hand to have more information

Thijxx, 2017-05-04:
Thanks, that's a great idea, to run in manually. Reading the code and running the script in debug mode made it clear to me.

Python script
`
21:22:19 [ERROR]: No response from device
`

ESP
```
Ready
IP address: 192.168.178.178
Start
Progress: 0%
Error[2]: Connect Failed
Error[4]: End Failed
```

The problem is that to upload the code a local service is created and it runs at a random port, apperantly this drove my UFW (Uncomplicated Firewall) crazy.
Assuming not all users run Windows zero security, it may be nice to at least make the port static and to include in the comments of the OTA script that you need to re-configure your firewall for **every** use as long as the port is not static :)

Firewall rule added
```
python2.7 espota.py -i 192.168.178.178 -f /some/path/BasicOTA.ino.generic.bin -d
21:28:26 [DEBUG]: Options: {'esp_ip': '192.168.178.178', 'host_port': 25623, 'image': '/some/path/BasicOTA.ino.generic.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': False}
21:28:26 [INFO]: Starting on 0.0.0.0:25623
21:28:26 [INFO]: Upload size: 247584
21:28:26 [INFO]: Sending invitation to: 192.168.178.178
21:28:26 [INFO]: Waiting for device...
Uploading..........................................................................................................................................................................
21:28:31 [INFO]: Waiting for result...
21:28:31 [INFO]: Result: OK
```

mmaret, 2017-05-05:
It's commented in BasicOTA, but you can set the port statically by using
 `ArduinoOTA.setPort(8266);`.
But, you are right, it's some more comments could help. May be you can propose a PR with a modification in BasicOTA with something like
`else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed. Firewall Issue ?");`

Thijxx, 2017-05-05:
Well no that's not the port I mean, this port (8266 default) is the port that the ESP will listen on for OTA communication. But the computer sending the data to the ESP also opens a port, that is randomized in the Python script or can be set using the script manually.
So I could try to write a Firewall check in the Python script and hopefully that script can then alert the Arduino IDE that the outgoing local port is being blocked.

igrr, 2017-05-08:
`espota.py` script which performs the OTA has an option to set local port. You could probably customize the build script in platformio or platform.txt file in Arduino to pass this extra argument and force the script to use some fixed port number.

closed 2017-05-08 by igrr