esp8266/Arduino #3188: Missing items in the doc

hmiez, 2017-05-02:
Thanks for the great work on SPIFFS !
I am basically willing to **write binary data** (bytes, words, long etc) in SPIFFS, with the idea to keep it compact and avoid character representations.
Something like   

- void f.put(long xyz);
- long f.get();

I have no idea if this is possible or not. 
At the moment, it **seems** that only text is supported by SPIFFS.
I have found text primitives like readString etc in different examples found on the net, but these are not mentionned on the github doc, maybe because they are inherited from the stream.

Maybe it is worth to add some information in the documentation to make clear if SPIFFS supports binary or text only data.

Thanks in advance.

igrr, 2017-05-02:
SPIFFS `File` objects inherit methods from Arduino `Stream` class, so you can use `read` and `write` methods:

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h#L55

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h#L65

hmiez, 2017-05-06:
Exactly what I was looking for.
Thanks !

closed 2017-05-08 by igrr