esp8266/Arduino #3192: Latest UART buffers break Serial.swap()

riataman, 2017-05-03:
I have a NodeMCU, I was using Serial.swap() just fine with 2.3.0. After upgrading to master Serial.swap doesn't work anymore. 

If I backup the serial buffer changes it works again.

igrr, 2017-05-08:
By "doesn't work", you mean that the output is still sent to the original pins after `swap`? Or some other issue?

igrr, 2017-05-18:
@riataman  Can you confirm that the issue is that you can not `swap` back to the original pins?

riataman, 2017-05-18:
Sorry, I haven't been able to get back to test this, but as I remember I was just calling ´Serial.swap´ once and the alternative pins didn't worked.

I'll try to get back to test this again soon.

devyte, 2017-12-28:
@riataman ping

d-a-v, 2018-10-12:
Closing due to age and because `Serial.swap()` is working.
Please open a new issue with more details if your tests are not satisfactory.

closed 2018-10-12 by d-a-v