esp8266/Arduino #3193: Connection refused with minimal GET sketch

anderspitman, 2017-05-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12S
Core Version:      	2.3.0

### Description

Wifi seems to be working great. I've verified that the 8266 is getting an IP and I am able to ping it from other machines on the network. But any sort of GET request (both local and internet) gives me a connection refused error. I suspect I'm missing something obvious here.

Using arduino 1.8.2 on Arch linux

### Settings in IDE

Module:  			Wemos D1 R2 & Mini
Flash Size: 		4MB
CPU Frequency:		80Mhz

### Sketch

```cpp

/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("ssid", "password");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;
        http.setTimeout(2000);

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        //http.begin("http://192.168.1.12/test.html"); //HTTP
        http.begin("http://192.168.0.160", 5000);

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}

```

### Debug Messages

```
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
```

anderspitman, 2017-05-03:
I had 2 overlapping problems. The first was that apparently the HttpClient is very picky about the trailing slash, so I needed to do `http.GET("http://192.168.0.160/")` as opposed to `http.GET("http://192.168.0.160")`. The other is that using the version of `begin` where you choose the port doesn't seem to work. I had to do `http.GET("http://192.168.0.160:5000/")`. Does anyone have more info about that?

igrr, 2017-05-03:
HTTPClient has two (HTTP) begin methods:
```
    bool begin(String url);
    bool begin(String host, uint16_t port, String uri = "/");
```
For the first one, you need to specify the full URL. Without a trailing slash, that isn't a full URL...

For the second one, you need to specify host, port, and URI, e.g. `http.GET("192.168.0.160", 5000, "/")`;

anderspitman, 2017-05-03:
Ah ok so dropping the "http" would be necessary in the second case. That makes complete sense; thanks.

mattolenik, 2018-01-28:
A trailing slash is not a requirement for a valid URL.

igrr, 2018-01-28:
I was using https://tools.ietf.org/html/rfc1738#section-3.3 as reference, which does not mention that the slash between `host` and `url-path` is optional. Do you have some other reference that it is?

arijitvt, 2018-02-11:
I was trying  
‘’’
http.begin(“http://www.google.com/“)
int rc = http.Get();
‘’’
I was getting -1/connection refused error.

KhamkarSagar, 2018-03-02:
I had same problem.mine problem fixed by changing Access point IP .when I used default IP ESP8266 it worked for 10 minutes,later on It was contineously showing -1 then I decided to change the Access point IP of ESP8266 and that worked for me.

Server Code:
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
 
IPAddress ip(192,168,1,200);   
IPAddress gateway(192,168,1,254);   
IPAddress subnet(255,255,255,0); 

const char* ssid = "ESPWebServer2157";
const char* password = "123456789";

ESP8266WebServer server(80); //Server on port 80

void handleRoot()
{
  server.send(200, "text/plain", "Welcome to ESP8266 Server:-Sagar Khamkar");
}

void setup(void){
    Serial.begin(115200);
  Serial.println(" ");
  //WiFi.config(lo);
  WiFi.softAPConfig(ip, gateway, subnet);
  WiFi.mode(WIFI_AP);           //Only Access point
  WiFi.softAP(ssid, password);  //Start HOTspot removing password will disable security
 
  IPAddress myIP = WiFi.softAPIP(); //Get IP address
  Serial.print("HotSpt IP:");
  Serial.println(myIP);
 
  server.on("/", handleRoot);  //Which routine to handle at root location
  //server.on("/", handleRoot);  //Which routine to handle at root location
  server.begin();                  //Start server
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();          //Handle client requests
}

Client Code:

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
const char* ssid = "ESPWebServer2157";
const char* password = "123456789";
HTTPClient http;
void setup () 
{

  Serial.begin(115200);
  WiFi.begin(ssid, password);
  delay(1000);
 Serial.print("Connecting to Access point..");
  while (WiFi.status() != WL_CONNECTED) 
  {
    Serial.print(".");
    delay(100);

  }
  delay(100);
  Serial.println("connected");

}

void loop() {

  if (WiFi.status() == WL_CONNECTED)
  { //Check WiFi connection status
  //Declare an object of class HTTPClient
     http.begin("http://192.168.1.200/");  //Specify request destination
     int httpCode = http.GET();  
    Serial.println(httpCode);
    //Send the request 
    if (httpCode > 0) { //Check the returning code

      String payload = http.getString();   //Get the request response payload
      Serial.println(payload);                     //Print the response payload

    }

    http.end();   //Close connection

  }

  delay(10000);    //Send a request every 10 seconds

}

manishsasmal, 2018-03-15:
i am facing the same issue. I think it might a problem with server. I am using a shared hosting from vultr.com i am getting error 400 Bad request. Server could not understand your request. but when i am using a shared hosting from others it is working perfectly. If you find any solution please post here. my code is as below

`#include <ESP8266WiFi.h>
#include <SPI.h>
#include "MFRC522.h"

int frequency=1000; //Specified in Hz
int buzzPin=0; //D3
int ledPin=2; //D4

#define RST_PIN 5
#define SS_PIN  4
MFRC522 rfid(SS_PIN, RST_PIN);  
unsigned int hex_num;

const char* ssid = "ssid";
const char* password = "password";
const char* host = "hostname";

void setup() {
  Serial.begin(9600);    // Initialize serial communications
  SPI.begin();           // Init SPI bus
  rfid.PCD_Init();    // Init MFRC522 
  pinMode(ledPin,OUTPUT);
  pinMode(buzzPin,OUTPUT);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
    {
      delay(500);
    }
  
}

void loop() { 
  if(WiFi.status() == WL_CONNECTED)
  {
    digitalWrite(ledPin, HIGH);
  }
  else
  {
    digitalWrite(ledPin, LOW);
  }
  // Look for new cards
  String rfid_esp="rfid=";
  if ((rfid.PICC_IsNewCardPresent()) & (rfid.PICC_ReadCardSerial()))
  {
    Serial.print(F("Card UID:"));
    hex_num =  rfid.uid.uidByte[0] << 24;
    hex_num += rfid.uid.uidByte[1] << 16;
    hex_num += rfid.uid.uidByte[2] <<  8;
    hex_num += rfid.uid.uidByte[3];
    long int  NFC_id=  (long int)hex_num;
    if(NFC_id < 0)
    {
      NFC_id = NFC_id * -1;
    }
    Serial.println(NFC_id);
    /*for (byte i = 0; i < rfid.uid.size; i++)
    {
      Serial.println(rfid.uid.uidByte[i]);
      rfid_esp += rfid.uid.uidByte[i];
      if(i<3)
      rfid_esp += ".";
    }
    Serial.println(rfid_esp);
    */
    //hit cloud api
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) 
    {
      Serial.println('not connected');
     // return;
      
    }
    String url = "project/register/";
    url += String(NFC_id);
    Serial.println(url);
    
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
    tone(buzzPin, frequency);
   delay(1000);
    noTone(buzzPin);
    while(client.available())
    {
      String line = client.readStringUntil('\r');
      Serial.print(line);
    }
  }
  
}
`

saumavadas, 2018-11-29:
Hello Guys, I am getting the same issue. Some times I got the 200 as well, Any one have the idea how to get rid off that problem.

 
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>

#define WIFI_SSID "****"
#define WIFI_PASSWORD "******" 

#define Relay1 D0
//int val1;

void setup() 
{
  Serial.begin(9600);
  pinMode(Relay1,OUTPUT);
  //digitalWrite(Relay1,HIGH);
  //pinMode(LED_BUILTIN, OUTPUT);
  
  WiFi.begin(WIFI_SSID,WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status()!=WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());
}

void loop() 
{
  //Serial.println("On Loop");
  if (WiFi.status() == WL_CONNECTED) 
  { 
    //Serial.println(WiFi.status());
    HTTPClient http;
    http.begin("http://iitnetsolutions.com/medinipumps/houseautomation.php?action=status"); 
    int httpCode = http.GET(); 
    Serial.println(httpCode);                                
    if (httpCode > 0) 
    { 
      //String PL =  http.getString();
      //Serial.print(PL);
      const size_t bufferSize = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(8) + 370;
      DynamicJsonBuffer jsonBuffer(bufferSize);
      JsonObject& root = jsonBuffer.parseObject(http.getString());
      String Switch_status = root["switch_status"];
      if(Switch_status == "1")                     
      {
        digitalWrite(Relay1,HIGH);
        Serial.println("ON");
        digitalWrite(LED_BUILTIN, HIGH);
      }
      else if(Switch_status == "0")                                                     
      {                                      
        digitalWrite(Relay1,LOW);
        Serial.println("OFF");
        digitalWrite(LED_BUILTIN, LOW);
      }    
      Serial.println(Switch_status);
    }
    http.end();   
  }
  delay(100);
[image]

}

d-a-v, 2018-11-29:
Can you please try latest git version ?
This is not mandatory but you would have to modify
```
HTTPClient http;
http.begin("http://iitnetsolutions.com/medinipumps/houseautomation.php?action=status"); 
```
to
```
HTTPClient http;
WiFiClient cli;
http.begin(cli, "http://iitnetsolutions.com/medinipumps/houseautomation.php?action=status"); 
```

saumavadas, 2018-11-29:
I have got this error:
error: no matching function for call to 'HTTPClient::begin(WiFiClient&, const char [73])'
http.begin(cli, "http://iitnetsolutions.com/medinipumps/houseautomation.php?action=status"); 
bool begin(String url);
HTTPClient\src/ESP8266HTTPClient.h:136:10: note:   candidate expects 1 argument, 2 provided
HTTPClient\src/ESP8266HTTPClient.h:137:10: note: bool HTTPClient::begin(String, String)
bool begin(String url, String httpsFingerprint);
HTTPClient\src/ESP8266HTTPClient.h:137:10: note:   no known conversion for argument 1 from 'WiFiClient' to 'String'
HTTPClient\src/ESP8266HTTPClient.h:138:10: note: bool HTTPClient::begin(String, uint16_t, String)
bool begin(String host, uint16_t port, String uri = "/");
HTTPClient\src/ESP8266HTTPClient.h:138:10: note:   no known conversion for argument 1 from 'WiFiClient' to 'String'
HTTPClient\src/ESP8266HTTPClient.h:139:10: note: bool HTTPClient::begin(String, uint16_t, String, String)
bool begin(String host, uint16_t port, String uri, String httpsFingerprint);

d-a-v, 2018-11-29:
That will work only with latest git version (= You need to update to "stage" version of the arduino git core).
I'm asking because we need to know if your bug is still present in current code.

saumavadas, 2018-11-29:
I not using the GIT, I am doing code on the Arduino IDE. I don't have the idea how to update the git latest version please help

d-a-v, 2018-11-29:
Or better:
https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

saumavadas, 2018-11-29:
ok let me check

saumavadas, 2018-11-29:
just for my knowledge, if I do not install the git what will be my issue?

d-a-v, 2018-11-29:
You have 3 options
- test with git so we can try and fix if the issue is still there
- stay with what you have, the issue may or may not be corrected in next release
- try to locate and fix the issue yourself with what you currently have

There is not enough manpower for another development model than a single branch.

ingitkumar, 2018-12-02:
I was getting "Connection Refused" error consistently. But in my case, I was doing a HTTP GET from inside of a Ticker function. Upon going through the library file, I found that the ClientContext library makes use of delay() function, which it seems, was not happy to work from inside of Ticker. In a hard way though, I learnt the lesson and so moved my function inside of loop() and used the ticker to enable a flag that would let my function to run from inside of loop(). Now the HTTP GET works happily with both the new and the older versions of the library.

fischerjade, 2019-04-10:
I don't know if this helps anyone but I had the same issue with my ESP8266 sketch and I just needed to update the fingerprint in the http.begin() argument.
EX:
http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"
To update this just go to https://www.grc.com/fingerprints.htm and add the full url and use the code as the second argument in the http.begin() method.

Fires04, 2019-04-18:
Hi as @ingitkumar  mention, it is not working in external or interrupts functions. I had to also move it to the main and make the flag variable to run it.

Shaileshverma123, 2019-06-10:
Hi all I am also getting similar error..HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused
can anyone help me out
@igrr

OttoKlaasen, 2019-06-23:
> I don't know if this helps anyone but I had the same issue with my ESP8266 sketch and I just needed to update the fingerprint in the http.begin() argument.
> EX: [...]

Nice , that worked for me

smtpsantiago, 2019-07-02:
this work for me, i try client for default

#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <HTTPClient.h>
#include <WiFiClientSecure.h>

WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("wifi", "password");

  // wait for WiFi connection
  Serial.print("Waiting for WiFi to connect...");
  while ((WiFiMulti.run() != WL_CONNECTED)) {
    Serial.print(".");
  }
  Serial.println(" connected");
}

void loop() {
  WiFiClientSecure *client = new WiFiClientSecure;
  HTTPClient https;

        Serial.print("[HTTPS] begin...\n");
  if (https.begin(*client, "https://hostname.com")) {  // HTTPS
        Serial.print("[HTTPS] GET...\n");
    int httpCode = https.GET(); 
    if (httpCode > 0) {
      Serial.printf("[HTTPS] GET... code: %d\n", httpCode);
      if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
        String payload = https.getString();
        Serial.println(payload);
      }
    } else {
      Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
    }

    https.end();
  } else {
    Serial.printf("[HTTPS] Unable to connect\n");
  }

  delete client;

  Serial.println();
  Serial.println("Waiting 10s before the next round...");
  delay(10000);
}

earlephilhower, 2020-01-01:
This is a long and winding issue with multiple separate issues and solutions posted to it.  From what I can see, though, with the current GIT, this is reported as working and there's nothing to do here.  If someone has a specific issue against the current release/GIT head, please open a new issue so it's trackable and actionable.

closed 2020-01-01 by earlephilhower

elricho19, 2020-01-24:
> this work for me, i try client for default
>
> #include <Arduino.h>
> #include <WiFi.h>
> #include <WiFiMulti.h>
> #include <HTTPClient.h>
> #include <WiFiClientSecure.h>
>
> WiFiMulti WiFiMulti;
>
> void setup() { [...]

Santiago, I try on my esp8266 but doesn't work, you use this code on a esp8266? The librarys are for this board?

sandorbedo, 2020-06-05:
I have the same issue.
I get "connection refused" error when I call the "GET()" method.
In my case I have an AP on ESP32 (I have tested it with browser and it works fine).
My client runs on ESP8266. (I have tested it with another public address and it works).
But when I connect the ESP8266 (client) to the ESP32 (server) every time I get this error.
I have already changed the IP address of them as mentioned before, but it did not help.
Do you have any another idea how can I fix this issue?

Server code:
`#include "WiFi.h"
#include "ESPAsyncWebServer.h"

const char* pSsid      = "ESP32-AP";
const char* pPassword  = "123456789";

const byte builtInLED = 2;

AsyncWebServer server(80);

IPAddress local_ip(192,168,1,1);
IPAddress gateway(192,168,1,254);
IPAddress subnet(255,255,255,0);

//ClosedCube::Sensor::TMP116 tmp116;

String readTemp() {
  Serial.println("Read temp called\n\r");
  return String(1.8);
}

String readHumi() {
  Serial.println("Read humidity called\n\r");
  return String(16.6);
}

String readPres() {
  Serial.println("Read pressure called\n\r");
  return String(33);
}

void setup()
{
  /* Serial port init */
  Serial.begin(115200);
  Serial.println();

  pinMode(builtInLED, OUTPUT);

  /* WiFi access point init */
  Serial.println("Setting AP...");
  boolean result = WiFi.softAP(pSsid, pPassword);
  WiFi.softAPConfig(local_ip, gateway, subnet);

  if(true == result)
  {
    Serial.print("WiFi soft access point is READY\n");
    Serial.print("IP:  "); 
    Serial.println(WiFi.softAPIP());
    Serial.print("MAC: "); 
    Serial.println(WiFi.softAPmacAddress());
  
    server.on("/temperature", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send_P(200, "text/plain", "HelloWorld");
    });
    
    server.on("/humidity", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send_P(200, "text/plain", readHumi().c_str());
    });
    
    server.on("/pressure", HTTP_GET, [](AsyncWebServerRequest *request){
[... 132 lines cut ...]
08:57:41.267 -> WiFi soft access point is READY
08:57:41.267 -> IP:  192.168.1.1
08:57:41.267 -> MAC: 30:AE:A4:90:0F:31

Client:
Connecting
09:09:17.182 -> .......
09:09:21.029 -> Connected
09:09:21.029 -> LocalIP:	192.168.1.200
09:09:21.029 -> MAC:		60:01:94:70:83:A8
09:09:21.029 -> Gateway:	192.168.1.254
09:09:21.029 -> AP MAC:		30:AE:A4:90:0F:31
09:09:21.553 -> Selected server: http://192.168.4.1/temperature/
09:09:26.690 -> 
09:09:26.690 -> [HTTP] GET... failed, error: connection refused
09:09:26.690 -> Temperature: --
09:09:26.690 -> Selected server: http://192.168.4.1/humidity
09:09:31.831 -> 
09:09:31.831 -> [HTTP] GET... failed, error: connection refused
09:09:31.831 -> Humidity: --
09:09:31.831 -> Selected server: http://192.168.4.1/pressure/
09:09:36.984 -> 
09:09:36.984 -> [HTTP] GET... failed, error: connection refused
09:09:36.984 -> Pressure: --
09:09:36.984 -> Temperature: -- *C - Humidity: -- % - Pressure: -- hPa

tophat17, 2020-11-25:
Oh, man... I'm still having this issue. I hope someone thinks of something soon before I lose my mind.

fischerjade, 2020-11-25:
I ended up not connecting this way.  I am using fred.sensetecnic.com.  I use the Node Red and MQTT that they provide.  They have instructions on how to use this on the fred.sensetecnic.com website.   
Jade Fischer

Ardutec, 2021-02-09:
> I don't know if this helps anyone but I had the same issue with my ESP8266 sketch and I just needed to update the fingerprint in the http.begin() argument.
EX:
http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"
To update this just go to https://www.grc.com/fingerprints.htm and add the full url and use the code as the second argument in the http.begin() method.

Thank-you, this worked for me too.

vlad2005, 2021-03-20:
> I was getting "Connection Refused" error consistently. But in my case, I was doing a HTTP GET from inside of a Ticker function. Upon going through the library file, I found that the ClientContext library makes use of delay() function, which it seems, was not happy to work from inside of Ticker. In a hard way though, I learnt the lesson and so moved my function inside of loop() and used the ticker to enable a flag that would let my function to run from inside of loop(). Now the HTTP GET works happily with both the new and the older versions of the library.
> 
> Your post helped me a lot @d-a-v . Thanks for mentioning it here.

U save my day!!!!
Million of thanks!
I can confirm, GET method for HTTTPClient not work when is call using Ticker