esp8266/Arduino #3195: Brief SoftAP Connection Loss During WiFi Connection Attempt

craigfoo, 2017-05-03:
### Basic Info

Hardware:			ESP-12S

### Description

When the SoftAP is active (the ESP8266 is acting as an Access Point), I attempt to connect to a WiFi router as a Station using WifiMulti.appAP("SSID", "PASS") then WifiMulti.run(). On successful connection, I notice that the SoftAP signal that is being broadcast is dropping out momentarily.

It's pretty hard to capture but I've been testing from my MacBook Air with https://packetsender.com/. 

### Order of Events:
1. I set WiFi.mode(WIFI_AP_STA). 
2. I fire up the SoftAP and start a UDP server with a hardcoded IP address and port. 
3. I then wait for a message from PacketSender with the SSID and Pw for the WiFi router.
4. I take those credentials and attempt the connection, this is where the SoftAP drops out. I know it's dropping out because I'm sending an acknowledge UDP packet back but the packet fails to send.

Has anyone else seen this behavior?

igrr, 2017-05-08:
Until SDK 2.0.0, WiFi stack didn't send channel switch announcements when connecting to an AP. This was implemented in SDK 2.0.0 (which is currently available in the master branch). However based on some reports (http://bbs.espressif.com/viewtopic.php?f=66&t=3328&p=11258, and another one on the nodemcu issue tracker which I can't find right now), this feature doesn't work properly with some devices.

craigfoo, 2017-05-08:
Thanks for link that post. I did some more digging after I posted this and found this post that pretty much answered it for me, http://bbs.espressif.com/viewtopic.php?f=10&t=324.

My work around right now is to do a network scan, find the channel of the strongest Wifi signal, and set the SoftAP to that channel before it's propped up. This might not work in a setting where there are several Wifi connections though. Is there a better work around that I'm missing?

devyte, 2017-09-06:
The channel hopping in the SoftAP is caused by the Station looking for the router (both share the same radio). There is no solution for this, just several workarounds, all of which are incomplete due to several reasons.
The channel change announcement seems to be bugged on espressif's side, due to an extra beacon being sent. If that is in fact true, it messes up the clients connected to the SoftAP.
However, it is unknown whether a fix for that will allow the clients connected to the SoftAP to follow the AP through the channel changes. In any case, there is currently no fix available for that by espressif, and there is also no ETA,
GIven that this is currently tagged, I won't close it for now, but all issues related to the channel hopping should be covered by a single issue.

devyte, 2018-02-28:
[This ](https://github.com/espressif/ESP8266_NONOS_SDK/issues/94)is relevant on Espressif's repo.