esp8266/Arduino #3196: sd card

schlaegerz, 2017-05-04:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

akaJes, 2017-05-10:
you can try updated library https://github.com/greiman/SdFat/ or https://github.com/greiman/SdFat-beta
i had various issues with embedded library also

devyte, 2017-11-24:
@schlaegerz which core version? What are the IDE settings? Ehat is the sketch?
Please attach a text file with the text content. Please post a minimal example to reproduce the issue.

schlaegerz, 2017-11-28:
Its been a long time since I posted this, so its hard to say. I believe this ended up being an issue of the library corrupting the sd card when it shut down in the middle of a write.

devyte, 2017-11-28:
@igrr I don't think there is much that can be done in this case. FAT file system corruption is always a risk with unexpected shutdowns.
Unless you had something specific in mind when you took a look, I propose closing this.

schlaegerz, 2017-11-28:
It would help to suggest using O_SYNC more often, that helped corrupt my file system significantly less. 

Also, it should be possible to do a bit better validation so that it doesn't just completely crash when the file system isn't completely right.

devyte, 2018-11-13:
This requires changes in the low level FAT lib, and I don't know if/when that will be done. Removing milestone in the meantime.

devyte, 2019-01-31:
Will be addressed, as far as possible, with the new SDFS.

earlephilhower, 2019-04-15:
Closing for now.  We've moved to a completely new SDFat library (and away from the old Arduino frozen copy of SD.h) and SPI is taking care of memory alignment as far as we've weel.  Open with a MCVE if you can reproduce with current GIT.

closed 2019-04-15 by earlephilhower