esp8266/Arduino #3198: Website down

NuclearCookie, 2017-05-04:
### Basic Infos

Host platform: Archlinux

### Description

http://arduino.esp8266.com appears offline. 
I cannot use http://arduino.esp8266.com/stable/package_esp8266com_index.json to add it to my build manager.
I also cannot build from source becaues the dependency http://arduino.esp8266.com/stable/package_esp8266com_index.json is also offline. 

According to IsItDown, site has been offline for 3 days.

http://www.isitdownrightnow.com/arduino.esp8266.com.html

Please fix or provide alternatives :) Thanks!

igrr, 2017-05-04:
That's an interesting one. I can open arduino.esp8266.com both in browser and in in command line with curl. However if i connect to a VPN in Japan, i can't resolve the IP.

@NuclearCookie can you check if opening http://download.igrr.me/stable/package_esp8266com_index.json in a browser works for you?

If it works, then likely there is some issue with esp8266.com DNS. If it doesn't, then it's some other issue.

NuclearCookie, 2017-05-04:
Hi @igrr,

Thanks for the swift response. 
The link you sent works fine in my browser.

I thought at first that the adress might not have been available behind my firewall, but I tried on my phone using cellular data and even then I can't access the page. (I'm in Belgium).

Any suggestion?

igrr, 2017-05-04:
I guess that's some DNS issue with arduino.esp8266.com record. I don't have access to this configuration though, maybe @themindfactory can check?

beicnet, 2017-05-04:
Yes, same here "This site can’t be reached",...

Regards

per1234, 2017-05-04:
I get "Server not found" for http://arduino.esp8266.com/ and http://arduino.esp8266.com/stable/package_esp8266com_index.json
I'm located in the United States

I get an error in the Arduino IDE:
```
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json
```
whenever this happens and I first noticed it a couple hours ago. I'd be surprised I hadn't seen it earlier if it's really been 3 days.

nazgee, 2017-05-04:
same here (Poland)
if that's the DNS issue, can anyone provide a server's IP address?

wolter, 2017-05-04:
104.131.82.128 seems to work - unfortunately this not sufficient. hence I use the github version and even there you need to replace the url in the package file inside the package folder with this ip. but this seems to do the trick... cheers, sascha.

igrr, 2017-05-04:
Based on the bandwidth graphs, i think DNS only went down today (also i got the monitoring notifications a few minutes before this issue was opened).
[image]

NuclearCookie, 2017-05-04:
@wolter How did you manage to do that? If I replace the http://arduino.esp8266.com by the IP, I get errors from python that the requiest is an unknown url type
```
raise ValueError("unknown url type: %r" % self.full_url)
ValueError: unknown url type: '104.131.82.128/linux64-xtensa-lx106-elf-gb404fb9.tar.gz'
```

I kind of need this today so even if it's through a hack, I'd be super happy :)

ScruffR, 2017-05-04:
@NuclearCookie , this link works for me
 http://download.igrr.me/stable/package_esp8266com_index.json

NuclearCookie, 2017-05-04:
Hi @ScruffR, That works indeed, but it doesn't fix the problem because that files just references to files on the arduino.esp8266 website. It's not a full mirror of the repository :(

wolter, 2017-05-04:
@NuclearCookie : I guess, you removed the http? I attached my package_esp8266com_index.template.json (folder Arduino\hardware\esp8266com\esp8266\package) for your convenience (
[package_esp8266com_index.template.json.txt](https://github.com/esp8266/Arduino/files/976192/package_esp8266com_index.template.json.txt)
). Hopes, it helps. Unfortunately this is not a generic solution, because IPs can change...

wolter, 2017-05-04:
btw these are the instructions how to use the github version: https://github.com/esp8266/Arduino#using-git-version - with my fix you get finally rid of the DNS error... cheers, sascha.

NuclearCookie, 2017-05-04:
That works like a charm. Thanks for helping out with this temporary solution!

ScruffR, 2017-05-04:
The git instructions didn't work for me on Windows 8.1Pro 64bit 😩

wolter, 2017-05-04:
@ScruffR make sure you have python installed and you use/create the correct folder structure.

ScruffR, 2017-05-04:
@wolter, I think I have followed the instructions correctly.
This is my error log
```
C:\Program Files (x86)\Arduino\hardware\arduino\esp8266com\esp8266\tools>python --version
Python 2.7.10

C:\Program Files (x86)\Arduino\hardware\arduino\esp8266com\esp8266\tools>python get.py
Platform: i686-mingw32
Tool esptool-0.4.9-win32.zip already downloaded
Extracting dist/esptool-0.4.9-win32.zip
Renaming esptool-0.4.9-win32/ to esptool
Downloading win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
Traceback (most recent call last):
  File "get.py", line 116, in <module>
    get_tool(tool)
  File "get.py", line 75, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "C:\Python27\lib\urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "C:\Python27\lib\urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "C:\Python27\lib\urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "C:\Python27\lib\urllib.py", line 350, in open_http
    h.endheaders(data)
  File "C:\Python27\lib\httplib.py", line 1049, in endheaders
    self._send_output(message_body)
  File "C:\Python27\lib\httplib.py", line 893, in _send_output
    self.send(msg)
  File "C:\Python27\lib\httplib.py", line 855, in send
    self.connect()
  File "C:\Python27\lib\httplib.py", line 832, in connect
    self.timeout, self.source_address)
  File "C:\Python27\lib\socket.py", line 557, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
IOError: [Errno socket error] [Errno 11001] getaddrinfo failed

C:\Program Files (x86)\Arduino\hardware\arduino\esp8266com\esp8266\tools>
```

wolter, 2017-05-04:
@ScruffR seems like you missed to change the file package_esp8266com_index.template.json (folder Arduino\hardware\esp8266com\esp8266\package). I attached it in one of my previous posts. Hope, it helps.

ScruffR, 2017-05-04:
In deed, I missed that, sorry 😳 
Thanks

jonpushkolli, 2017-05-04:
i cant download from the link i keep getting an error how do I fix?

JoranJix, 2017-05-04:
Used the workaround with git on Win10, but now nothing shows up in the boardmanager. 
https://gyazo.com/73b28aab1efb3d40ccbf2794e3c0c9fc

frontier777, 2017-05-04:
Thank you for the IP address above, I just added a local hosts file entry and my download is now working.
104.131.82.128	arduino.esp8266.com

sebastiankliem, 2017-05-04:
Same here (Germany). Using the IP address instead of http://arduino.esp8266.com works for me.

JoranJix, 2017-05-04:
created a temporary package to install it on my system... so if anyone like to use it
http://diy.mikolex.de/package_esp8266com_index.json

anik786, 2017-05-04:
Thanks @JoranJix . Using your link till it gets fixed.

jonpushkolli, 2017-05-04:
how to i fix this?

anik786, 2017-05-04:
@jonpushkolli Just use the link @JoranJix  gave

chivingtoninc, 2017-05-04:
Yup I'm in the US and site is down here too

chivingtoninc, 2017-05-04:
Also still getting this:

"linux64-xtensa-lx106-elf-gb404fb9.tar.gz"

Even with replacing the ip in the package file. 

Anyone help?

JoukeD, 2017-05-04:
chivingtoninc you can use the package json provided by wolter a couple of comments up. Works like a charm!

JoukeD, 2017-05-04:
@wolter Thanks for your json file!, works like a charm!

loquendowolf, 2017-05-04:
error Error descargando http://arduino.esp8266.com/stable/package_esp8266com_index.json
 Error descargando http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
java.lang.RuntimeException: java.lang.Exception: Error descargando http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error descargando http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.net.UnknownHostException: arduino.esp8266.com
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more
 Error descargando https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip
java.lang.RuntimeException: java.lang.Exception: Error descargando https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error descargando https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)

scrodev, 2017-05-04:
Its just a DNS Problem. Server is working great.
Workaround until the problem is fixed:
add this in hosts file
`104.131.82.128  arduino.esp8266.com`
then use another json file. For example: https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json

4dsystems, 2017-05-05:
Effecting Travis CI, Pull request is failing. I guess we wait for a solution?

dogrocker, 2017-05-05:
Change hosts file as @scrodev suggest. But it not work when downloading from arduino IDE.
`Error downloading http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz`
In my browser when I browse to `http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz` It show `Service unavailable`.

4dsystems, 2017-05-05:
Not quite sure how changing my local hosts file is going to help Travel CI accept my pull request to the repo... Trying to add changes. If that is what you are meaning?

NuclearCookie, 2017-05-05:
Hi @dogrocker  Please scroll back in this thread and check the comment from @wolter. It has some upvotes. There are plenty of solutions in this thread already.

loquendowolf, 2017-05-05:
Error htpp://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz appears in boards manager

loquendowolf, 2017-05-05:
If I download the library directly
Where to copy that file into the arduino folder

shariramani, 2017-05-05:
Tested working in win7 and win10 by adding hosts entry

Go to Start > run Notepad.
Right click on the Notepad icon and select Run as a administrator.
Select Open from the File menu option.
Browse to c:\windows\system32\drivers\etc\hosts. (\etc folder may be hidden, if its not shown just type file path )
Add the host name and IP address (as shown in following line) to the bottom of the host file.
104.131.82.128 arduino.esp8266.com
Save the host file.

You are all set to go. Original link  http://arduino.esp8266.com/stable/package_esp8266com_index.json will work now.

igrr, 2017-05-05:
@themindfactory arduino.esp8266.com name resolves now, but points to esp8266.com instead.

igrr, 2017-05-05:
Yay seems like DNS is fixed!

vicnevicne, 2017-05-05:
Suggestion: why not use the github copy as the official board manager link ? Repeated issues #1689 #1951 #2664 #3035 show that relying on a custom domain (even though it's owned by Espressif) may not be a very good idea... Of course, it will require a migration which probably means uninstalling and reinstalling, but that may be worth to hassle to get rid of these issues once and for all... Just my 2 cents.

matpalm, 2017-05-05:
and down again.. :(

```
$ curl -I arduino.esp8266.com/stable/package_esp8266com_index.json
HTTP/1.1 521 Origin Down
Date: Fri, 05 May 2017 19:16:00 GMT
Content-Type: text/html; charset=UTF-8
Connection: keep-alive
Set-Cookie: __cfduid=daf2bda7d42957bfa6dd3928c931bfb231494011760; expires=Sat, 05-May-18 19:16:00 GMT; path=/; domain=.esp8266.com; HttpOnly
Expires: Thu, 01 Jan 1970 00:00:01 GMT
Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0
Pragma: no-cache
X-Frame-Options: SAMEORIGIN
Server: cloudflare-nginx
CF-RAY: 35a6081ee3aa284c-SJC
```

loquendowolf, 2017-05-05:
I change host file with new 104.131.82.128   arduino.esp8266.com  but error still appear.

loquendowolf, 2017-05-05:
# Copyright (c) 1993-2009 Microsoft Corp.
#
# This is a sample HOSTS file used by Microsoft TCP/IP for Windows.
#
# This file contains the mappings of IP addresses to host names. Each
# entry should be kept on an individual line. The IP address should
# be placed in the first column followed by the corresponding host name.
# The IP address and the host name should be separated by at least one
# space.
#
# Additionally, comments (such as these) may be inserted on individual
# lines or following the machine name denoted by a '#' symbol.
#
# For example:
#

#      102.54.94.97     rhino.acme.com          # source server
#       38.25.63.10     x.acme.com              # x client host

# localhost name resolution is handled within DNS itself.
#	127.0.0.1       localhost
#	::1             localhost
# 104.131.82.128   arduino.esp8266.com

devyte, 2017-05-05:
Down for me as well (Chile), both from windows 7 and android.

jayniz, 2017-05-05:
Down for me as well (Germany), `http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz`, `http://104.131.82.128/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz/` and its variants not available. 

@igrr let me know if you need some help/sparring for the hosting.

Does anybody else happen to have `osx-xtensa-lx106-elf-gb404fb9-2.tar.gz`?

ilianiv, 2017-05-05:
Error downloading http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz, too :(

http://arduino.esp8266.com responds with Cloudflare's host down page.

majenkotech, 2017-05-05:
Ditto here. Cloudflare tells me a host error:

> The web server reported a bad gateway error.

I'm just uploading what files I have to my web server. It's taking forever over my uberslow connection...

* http://majenko.co.uk/esp8266/

paulmassen, 2017-05-05:
Great! Waiting for the windows version! 

Others might want to change their json inside their arduino/hardware folder and update as follow for retrieving it from your server if that's okay for you

```
{
  "packages": [
    {
      "maintainer": "ESP8266 Community", 
      "help": {
        "online": "http://esp8266.com/arduino"
      }, 
      "websiteURL": "https://github.com/esp8266/Arduino", 
      "platforms": [
        {
          "category": "ESP8266", 
          "name": "esp8266", 
          "url": "https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip", 
          "checksum": "SHA-256:884188a54bc75f1f2e7f827619f8cb1980dc3f430f92f9b07186ed7f4db8c738", 
          "help": {
            "online": "http://esp8266.github.com/Arduino/versions/2.3.0/"
          }, 
          "version": "2.3.0", 
          "architecture": "esp8266", 
          "archiveFileName": "esp8266-2.3.0.zip", 
          "boards": [
            {
              "name": "Generic ESP8266 Module"
            }, 
            {
              "name": "Olimex MOD-WIFI-ESP8266(-DEV)"
            }, 
            {
              "name": "NodeMCU 0.9 (ESP-12 Module)"
            }, 
            {
              "name": "NodeMCU 1.0 (ESP-12E Module)"
            }, 
            {
              "name": "Adafruit HUZZAH ESP8266 (ESP-12)"
            }, 
            {
              "name": "ESPresso Lite 1.0"
            }, 
            {
              "name": "ESPresso Lite 2.0"
            }, 
            {
              "name": "Phoenix 1.0"
            }, 
            {
              "name": "Phoenix 2.0"
            }, 
            {
              "name": "SparkFun Thing"
            }, 
            {
              "name": "SweetPea ESP-210"
            }, 
            {
              "name": "WeMos D1"
            }, 
            {
              "name": "WeMos D1 mini"
            }, 
            {
              "name": "ESPino (ESP-12 Module)"
            }, 
            {
              "name": "ESPino (WROOM-02 Module)"
[... 584 lines cut ...]
              "size": "46281"
            }, 
            {
              "url": "https://github.com/igrr/mkspiffs/releases/download/0.1.2/mkspiffs-0.1.2-linux32.tar.gz", 
              "checksum": "SHA-256:e990d545dfcae308aabaac5fa9e1db734cc2b08167969e7eedac88bd0839667c", 
              "host": "i686-pc-linux-gnu", 
              "archiveFileName": "mkspiffs-0.1.2-linux32.tar.gz", 
              "size": "45272"
            }, 
            {
              "url": "https://github.com/igrr/mkspiffs/releases/download/0.1.2/mkspiffs-0.1.2-linux-armhf.tar.gz", 
              "checksum": "SHA-256:5a8836932cd24325d69054cebdd46359eba02919ffaa87b130c54acfecc13f46", 
              "host": "arm-linux-gnueabihf", 
              "archiveFileName": "mkspiffs-0.1.2-linux-armhf.tar.gz", 
              "size": "41685"
            }
          ]
        }
      ], 
      "email": "ivan@esp8266.com", 
      "name": "esp8266"
    }
  ]
}
```

majenkotech, 2017-05-05:
Windows is (as always) the big one... 28% uploaded so far... :/

majenkotech, 2017-05-05:
Ok, uploads done. Unfortunately I don't have the ARM Linux version of the compiler.

illukas, 2017-05-05:
Thanks majenkotech.
Was able to install everything from your server, using your .json and Arduino IDE 1.8.1;

soneric1982, 2017-05-05:
Thanks majenkotech.

Is working http://majenko.co.uk/esp8266/package_esp8266com_index.json

soneric1982, 2017-05-05:
Thanks @majenkotech 
Is working http://majenko.co.uk/esp8266/package_esp8266com_index.json

carlosvolt, 2017-05-06:
In Argentina no working

igrr, 2017-05-06:
Fixed again :)

@vicnevicne I am all for hosting this on Github (the server is not affiliated with Espressif).

rrbautista, 2017-05-06:
The site is also down here in Philippines.

I used the link of @JoranJix above and successfully downloaded and installed the boards. Thanks.

Waiting for the permanent fix....

4dsystems, 2017-05-06:
http://arduino.esp8266.com/ shows 'no file specified' now when I load it in the browser, rather than page cannot be found/displayed etc. Is it back up?

[30 more replies]

closed 2017-05-06 by igrr