esp8266/Arduino #3204: Uploading new firmware which uses git version of this library via OTA removes WiFi credentials when the old firmware was using stable version

Defozo, 2017-05-05:
### Basic Infos

I have a firmware on my ESP-07S which was compiled using 2.1.0 version of this esp8266-arduino library.
This firmware has OTA functionality. I can update my firmware using my server.

#### Hardware
Hardware:			ESP-07S
Core Version:      	2.1.0 vs master

### Description

Everything is fine when I'm updating my firmware with the one compiled using 2.1.0 version of this library. However, when I try to use the latest version available by cloning git repository my WiFi credentials are being removed after OTA update - the device can't connect.

Is there a way to prevent this? Why is it happening?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB (1M SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		ck

devyte, 2017-09-06:
I think the removal of the on-flash credentials is on purpose, and that before they were being kept out of pure luck.
I believe that the SDK stores the credentials in a particular place on the flash, together with other pieces of data (e,g.; wifi RF calibration data). When a new firmware is flashed, especially when there is a new SDK at play, the place in the flash where the data is stored, as well as the layout of the data itself on the flash, could change, making your previous data invalid.
I could be wrong. @igrr?

I suggest not relying on those credentials anyways. They are meant for quick connection on reset or wakeup, to reduce reboot time. Personally, I don't even use them, and store all configuration data, including wifi credentials, in a file on SPIFFS, and leave persistent(false).

earlephilhower, 2019-01-31:
Closing as no longer relevant, relating to the 2.1 arduino release and the Espressif SDK release that came out a couple years ago.

closed 2019-01-31 by earlephilhower