esp8266/Arduino #3210: ESP refuses to accept code

Zeragonii, 2017-05-06:
### Basic Infos

Pretty much what it says in the title, my handful of ESP8266's refuse to accept code at all.

#### Hardware
Hardware:			ESP-01
Core Version:      	Not a clue how to even find this one out. 

### Description

Been trying for 3 days now to get code onto my ESP8266's, I have 7 of them now thinking it was a bad ebay batch causing my issues, but no matter what I try the module just will not accept code. I've got my wiring done as TX>RX:RX>TX:GND>GND:VCC>VCC:GP0>GND and that's it, no resistors or anything, I was following a video on youtube and mirrored his wiring exactly and still no results. The module powers on and I've checked the voltages on all connections and nothing exceeds 3.3v but whenever I try to flash code the LED onboard shines bright for half a second, then 2 small blinks, then bright again. Then I get my error messages which will be pasted below. I've even gone as far as getting a second USB to SERIAL module for fear that might be the fault, neither of them have yielded results for me.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:	80MHz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		USB to SERIAL
Reset Method:		ck

### Sketch

Blink example sketch

### Debug Messages

Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, none"

D:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Josh\AppData\Local\Arduino15\packages -tools D:\Program Files (x86)\Arduino\tools-builder -tools D:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Josh\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Josh\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Serial,DebugLevel=WiFic___ -ide-version=10801 -build-path C:\Users\Josh\AppData\Local\Temp\arduino_build_938299 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\Josh\AppData\Local\Temp\arduino_modified_sketch_974170\Blink.ino
D:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware D:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Josh\AppData\Local\Arduino15\packages -tools D:\Program Files (x86)\Arduino\tools-builder -tools D:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Josh\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Josh\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Serial,DebugLevel=WiFic___ -ide-version=10801 -build-path C:\Users\Josh\AppData\Local\Temp\arduino_build_938299 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\Josh\AppData\Local\Temp\arduino_modified_sketch_974170\Blink.ino
Using board 'generic' from platform in folder: C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Josh\AppData\Local\Temp\arduino_build_938299/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Josh\AppData\Local\Temp\arduino_build_938299/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\sketch\Blink.ino.cpp" -o "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Josh\AppData\Local\Temp\arduino_build_938299/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Josh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\sketch\Blink.ino.cpp" -o "C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\sketch\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using previously compiled file: C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\core\cont.S.o
Using previously compiled file: C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\core\cont_util.c.o
Using previously compiled file: C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\core\core_esp8266_eboot_command.c.o
Using previously compiled file: C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\core\core_esp8266_flash_utils.c.o
Using previously compiled file: C:\Users\Josh\AppData\Local\Temp\arduino_build_938299\core\core_esp8266_i2s.c.o
[... cut ...]
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

igrr, 2017-05-08:
Please follow this FAQ entry if the steps described there do not work, please explain what you did and what results you got.

devyte, 2017-09-07:
I hope you got your modules working. If you are still having trouble with your modules, try from a different vendor, or a different board flavor. 

Currently I don't believe there are issues with flashing. Having said that, this is not the correct place to ask for support to get your module working for the first time. This is an issue tracker, meant for tracking issues in the core. Please ask for assistance at esp8266.com or stackoverflow.
Closing.

closed 2017-09-07 by devyte