esp8266/Arduino #3211: POST Request from web page

mcmega, 2017-05-07:
Hello everyone!
Enable Web server on this core.
From the web page, you need to send the POST requests /dev.sc-write.
How can I take useful information from a query?

horendus, 2017-05-08:
You need to attach a POST method handler to the directory of where you are submitting the POST request.

server.on(/, "POST", processPostData());   <---This will handle post requests submitted to / or "root")

The POST handling method should then iterate through the POST requests arguments and be processed in what ever way you see fit. The names and values of the POST requests elements can be accessed with server.argName() and server.arg(). The total number of post argument that have been received is available through server.args() (a return value is given)

(Note the bellow is untested and uncomplied) 
```
void processPostData(){
 if (server.args() > 0){
    for (int i = 0; i < server.args();i++;){
        Serial.print("POST Arguments: " ); Serial.println(server.args(i));
        Serial.print("Name: "); Serial.println(server.argName(i));
        Serial.print("Value: "); Serial.println(server.arg(i));
}
```

Hope this helps.

mcmega, 2017-05-08:
Thank you!
I did this:

```
HTTP.on("/dev.sc-write", HTTP_POST, DevSCWrite);

void DevSCWrite() {
  if (HTTP.args()) {
    for (int i = 0; i < HTTP.args(); i++) {
      Serial.print("Name: "); Serial.println(HTTP.argName(i));
      Serial.print("Value: "); Serial.println(HTTP.arg(i));
    }
  }
  HTTP.send(200, "text/plain", "OK");
}
```
In UART is transmitted:
```
Name: ------WebKitFormBoundarysfF8OdaYVAF5BAJB
Content-Disposition: form-data; name
Value: "data"

NM:Name SC
ST:0
EV:[{RA:1001,EP:[e1_c1]}]
AC:[{RA:1001,AP:[a1_o1]}]
------WebKitFormBoundarysfF8OdaYVAF5BAJB--
```
How can I transfer only useful data to UART?
```
NM:Name SC
ST:0
EV:[{RA:1001,EP:[e1_c1]}]
AC:[{RA:1001,AP:[a1_o1]}]
```

igrr, 2017-05-08:
How are you sending the POST request? Could you post a code/webpage snippet?

mcmega, 2017-05-08:
I do this on nodejs + preact

```
postSend() {
		const data = 'Test Data: bla-bla';
		var body = new FormData;
		body.append('sc', data);
		fetch('dev.sc-write', {
			method: 'POST',
			body
		});
	}
```

mcmega, 2017-05-08:
Are you planning to remove this bug from the library Parsing.cpp and make case-insensitive headers?

igrr, 2017-05-08:
I think case-insensitive parsing was implemented in this PR: https://github.com/esp8266/Arduino/issues/2131

You can try using the master branch to see if it resolves the issue.

mcmega, 2017-05-08:
Thank you!
I did so, but I would like everything to work correctly from the package, without editing the author's library.
Why do not you want to add a few lines of code to the source?

igrr, 2017-05-08:
> Why do not you want to add a few lines of code to the source?

Could you clarify what you mean? The fix for the issue has already been merged into the master branch. When the next stable package is released, the fix is going to be included.

mcmega, 2017-05-08:
I meant all the HTTP headers, not just the Content-Type.

igrr, 2017-05-08:
Yes the PR mentioned above fixed parsing of other headers as well, not just the Content-Type:

https://github.com/esp8266/Arduino/commit/4897e0006b5b0123a2fa31f67b14a3fff65ce561

closed 2017-05-08 by igrr