esp8266/Arduino #3212: ESP8266mDNS - Add support for overwriting/removing TXT records

WeeJeWel, 2017-05-07:
I advertise a device's name on mDNS using `MDNS.addServiceTxt("myservice", "tcp", "name", "Foo");`, but this name can change over time by user input. When calling the method again, another TXT entry is added, but I expected the previous one to be replaced.

igrr, 2017-05-08:
@swarren and @ccooper21 have contributed to mDNS library recently. Maybe they would be willing to implement this?

Alternatively, you can get MDNS object into a clean state like this:
```c++
MDNS = MDNSResponder();
```
which should delete the old one, freeing all resources, and create a new one. Now if this doesn't work for some reason, feel free to open a new issue :)

mkillewald, 2017-12-07:
I realize this is probably low priority, but I would like to see support for being able to change or remove TXT records as well.  I could not get the suggestion of `MDNS = MDNSResponder()` to work. It clears the MDNS object but then I can't seem to re-initialize the object with either `MDNS.begin()`, `MDNS.setInstanceName()`, or `MDNS.addService()`.

romuye123, 2018-02-22:
@mkillewald 
Thanks for the UPDATE. But when I keep monitoring the updates on my Android app, the updates are not realtime / immediate. 

```
  MDNS.clearServiceTxt("http", "tcp");
  MDNS.addServiceTxt("http", "tcp", "name",myname); 
```
Please suggest if I am wrong or let me know how can I instantly update my client mobile app.

mkillewald, 2018-02-22:
@romuye123 
Due to the nature of mDNS the updates are not going to occur immediately. As a work around for this in my mobile clients, I've made a refresh action that drops all discovered nodes and re-discovers them causing the changes to be shown. Hope that helps!

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte