esp8266/Arduino #3213: Update the methods please!

treii28, 2017-05-08:
This is for the Arduino platform. The Arduino standard library includes a type String. Why do I still need to add a .c_str() to the end of all my parameters???

Humancell, 2017-05-08:
You don't have to add it to all of your parameters, but in this case the API that you referenced is expecting two pointers to char buffers.

The API is designed to be similar to the Arduino WiFi APIs ... which also take char buffers.

https://www.arduino.cc/en/Reference/WiFiBegin

igrr, 2017-05-08:
The latest official WiFi101 library does support String arguments:
https://github.com/arduino-libraries/WiFi101/blob/master/src/WiFi101.h#L116
so i suppose this is a valid enhancement request.

treii28, 2017-05-08:
but aren't most of the Arduino WiFi APIs built to talk to a connected ESP8266 module?

Humancell, 2017-05-08:
No.  The Arduino WiFi APIs were designed to support the Arduino WiFi shield.  You know that you can easily Google this type of information:  https://www.arduino.cc/en/Reference/WiFi

This project is very different:

_> This project brings support for ESP8266 chip to the Arduino environment. It lets you write sketches using familiar Arduino functions and libraries, and run them directly on ESP8266, no external microcontroller required.
> 
> ESP8266 Arduino core comes with libraries to communicate over WiFi using TCP and UDP, set up HTTP, mDNS, SSDP, and DNS servers, do OTA updates, use a file system in flash memory, work with SD cards, servos, SPI and I2C peripherals._

devyte, 2018-12-23:
Closing via #5295

closed 2018-12-23 by devyte