esp8266/Arduino #3215: Update to SDK 2.1.0

igrr, 2017-05-08:
non-OS SDK [2.1.0 has been released](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0), with support for 16 MB flash size, support for broadcast packets in ESP-NOW, fixes for ARP, UDP broadcast, and other issues.

- [ ] Update libraries
- [ ] Run existing tests
- [ ] Test 8/16 MB flash support
- [ ] Test new WiFi reconnect behaviour using WiFiManager library

alirezza, 2017-05-08:
Because of fixing ARP issue, It's important to upgrade to this SDK,
How long does it take to upgrade to this SDK ?

igrr, 2017-05-08:
@alirezza updating libraries should probably take a couple of hours (there's usually some breakage, SDK archives repackaging, and LD script hacking involved). As for the other parts of the checklist, depends, I wouldn't venture a guess.

Also I must mention that the top priority now is to clear all the blocking issues and push a new stable release, even if it doesn't include an SDK update. Feel free however to contribute to this issue if you find it important.

jonhp, 2017-05-08:
How about combining your top priority with the new SDK. Possibly one could help the other?

capedra, 2017-05-22:
@igrr How's it going? Could you please implement this for the next version 2.4.0-RC2 ? Thanks!

torntrousers, 2017-05-22:
+1 from me for getting the 2.1.0 SDK in the next release!

hcs-svn, 2017-05-22:
+1 because of the fixed ARP bug (#2330). To get rid of this would be very important.

igrr, 2017-05-23:
Created https://github.com/esp8266/Arduino/pull/3278.

Only ran webserver, PWM, and OTA tests. Folks interested in SDK 2.1.0 are encouraged to try that branch and report what works and what doesn't.

To use that PR, follow instructions for installing git version (remove the version installed using boards manager first). Then check out the PR by doing `git checkout update_sdk_2.1.0`.

hcs-svn, 2017-05-30:
> Folks interested in SDK 2.1.0 are encouraged to try that branch and report what works and what doesn't.

I have a sketch that uses:
- connect to an access point
- provide access point for configuration
- I2C (BMP180, BME180, LM75, MCP23008, SC16IS750, SSD1306 OLED)
- SPI (two RFM69)
- analog
- several WiFiServer where clients connect
- web frontend using ESP8266WebServer
- ota update using Update.begin, write, end ... in combination with ESP8266WebServer upload
- serial port communication

Everything work fine since several days.

TridentTD, 2017-08-01:
Next release please use   esp-mesh by   this newest version
https://github.com/espressif/ESP8266_MESH_DEMO    

esp-mesh on sdk2.1.0  still not update.

devyte, 2017-09-06:
@igrr what do you need to move forward with this? My knowledge of what is required is pretty much nullptr, but I'm willing to help out as I can starting this coming weekend.

igrr, 2017-09-06:
Thanks @devyte. Do any of the to-do items listed in the PR (https://github.com/esp8266/Arduino/pull/3278#issue-230659475) make sense to you? I think testing whether new SDK implementation of wifi reconnect works with WiFiManager is something you could help with.

devyte, 2017-09-06:
@igrr most don't. I'm sorry, I'm a higher level code developer, since decades, so my low level skills are beyond rusty. 
I think I can do the wifimanager test. Also, I've been toying with getting a few Wemos D1 mini pros. I'll buy them tomorrow for testing, but they'll take a (long) while to reach me.

igrr, 2017-09-06:
Also, i am not sure about merging SDK2.1.0 into 2.4.0. Pretty hard to say what might get broken. Should not prevent finishing work on SDK2.1.0 though...

igrr, 2017-09-06:
(also, probably OT here, but maybe some items in the "pending/todo" columns of https://github.com/esp8266/Arduino/projects/1 might make sense to you — once these are resolved, we can release 2.4.0 or at least -rc2).

devyte, 2017-09-26:
@igrr I just looked at the 2.1.0 changelist, and there are some rather important fixes in there, such as a fix for connecting to a wifi with hidden ssid, and the UDP broadcast extra bytes issue.
Also, there are 10 commits to the espressif github after the 2.1.0 release. Some of them seem to be fixes to outstanding issues, like the wpa2 enterprise identity api, and a fix for long time deep sleep. There are some iram optimizations as well.
Should we consider pulling the latest git NONOS? I think it merits at least exploring. If not for the release, then at least to be merged into master right after.

You said: 

> testing whether new SDK implementation of wifi reconnect works with WiFiManager

The only change in the reconnect behavior of the wifi mentioned in the changelist is when trying to connect to an AP that doesn't exist. I do see mentions of changes in the wifi events, though.
Should I just test whether the wifimanager works as expected?

igrr, 2017-09-26:
> Should we consider pulling the latest git NONOS?

Good point, I will update the changeset based on latest git of the SDK.

> Should I just test whether the wifimanager works as expected?

Yes, that would be great, thank you!

freedib, 2017-10-03:
I just looked at SDK 2.1 update because I needed new header definitions which were not in SDK 2.0.
It seems some file in https://github.com/esp8266/Arduino/tree/update_sdk_2.1.0/tools/sdk/include are the old ones or at least not the same as in ESP8266_NONOS_SDK.
I had concerns about in c_type.h (expected a definition for size_t) and uart_register.h (expected a definition for UART_TXD_INV).
Are there any reasons for this difference? Thanks

hemangjoshi37a, 2017-10-09:
does anyone know how to install this? I have downloaded the zip file but dont know how to install in my system. Please help me.

devyte, 2017-10-30:
@igrr I've been testing the WiFiManager examples for a while now, and tbh I don't really see any difference. The only issue I encountered is that the scan-related page loads fail sometimes. From what I can tell, it's due to one of the following:
- the WiFiManager object doesn't use async scanning, and so you face the channel hopping issue and related random disconnect from the SoftAP
- the ESP8266Webserver issue, or rather, the lwip1.4 issue caused by browsers doing multiple requests (same symptom I encountered a while back, which caused me to move to the ESPAsyncWebserver)

So I don't think the WiFiManager behavior is impacted, at least as far as I can tell. 
My conclusion for the WiFiManager: green light.

dalbert2, 2017-12-10:
I've been testing 2.1.0-rc2 WPS functionality to see if it fixed WPS with modern (802.11ac) routers and unfortunately it still does not work although it works perfectly with older 802.11b/g/n routers.
See issue: https://github.com/esp8266/Arduino/issues/3941

closed 2018-01-02 by igrr

JaimeSilva, 2018-03-24:
How can I upload to a new SDK? should I use ESP FLASH DOWNLOAD TOOL?