esp8266/Arduino #3216: xtensa error

Supersimo88, 2017-05-08:
#### Hardware
Hardware:			ESP8266

### Description

I'm getting this errors

c:/users/super/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'

c:/users/super/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':

e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

### Settings in IDE

Module:  			Generic ESP8266 module
Flash Size: 		1M (128K SPIFFS)
CPU Frequency:		?80Mhz?
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		HARDWARE

### Sketch

#include <ESPHelper.h>
//#include <HSBColor.h>

#include <my9291.h>

#include "math.h"

//#include <hsi2rgbw.h>

#define TOPIC "/light/rgb"
#define STATUS TOPIC "/status"
#define NETWORK_HOSTNAME "rgblight"
#define OTA_PASSWORD "ENOMIS55"

//#define RED_PIN D7    
//#define GREEN_PIN D6
//#define BLUE_PIN D5
//#define BLINK_PIN D4

#define MY9291_DI_PIN   13
#define MY9291_DCKI_PIN 15

my9291 _my9291 = my9291(MY9291_DI_PIN, MY9291_DCKI_PIN, MY9291_COMMAND_DEFAULT);

#define DEG_TO_RAD(X) (M_PI*(X)/180)

typedef struct lightState{
  double hue;
  double saturation;
  double brightness;
  int red;
  int redRate;
  int green;
  int greenRate;
  int blue;
  int blueRate;
  //int fadePeriod;
  //int updateType;
  int white;
  int whiteRate;
};

lightState nextState;

boolean newCommand = false;

char* lightTopic = TOPIC;
char* statusTopic = STATUS;
char* hostnameStr = NETWORK_HOSTNAME;

//const int redPin = RED_PIN;
//const int greenPin = GREEN_PIN;
//const int bluePin = BLUE_PIN;
//const int blinkPin = BLINK_PIN;

char statusString[50];  //string containing the current setting for the light

//set this info for your own network
netInfo homeNet = {.name = "rgblight", .mqtt = "192.168.1.19", .ssid = "Telecom-19250793", .pass = "yxsplyuvMmEMlS5vbg67MPK3"};

ESPHelper myESP(&homeNet);

void setup() {
  //initialize the light as an output and set to LOW (off)
  //pinMode(redPin, OUTPUT);
  //pinMode(bluePin, OUTPUT);
  //pinMode(greenPin, OUTPUT);

  //all off
  //digitalWrite(redPin, LOW);  //all off
  //digitalWrite(greenPin, LOW);
  //digitalWrite(bluePin, LOW);

_my9291.setColor((my9291_color_t) { 0, 0, 0, 255 }); // W 100% duty cycle
    _my9291.setState(true);

  delay(1000);

  
  //setup ota on esphelper
  myESP.OTA_enable();
  myESP.OTA_setPassword(OTA_PASSWORD);
  myESP.OTA_setHostnameWithVersion(hostnameStr);

  //myESP.enableHeartbeat(blinkPin);
  //subscribe to the lighttopic
  myESP.addSubscription(lightTopic);
  myESP.begin();
  myESP.setCallback(callback);

}
void loop() {

  static bool connected = false; //keeps track of connection state

  if(myESP.loop() == FULL_CONNECTION){

    //if the light was previously not connected to wifi and mqtt, update the status topic with the light being off
    if(!connected){
      connected = true; //we have reconnected so now we dont need to flag the setting anymore
      myESP.publish(statusTopic, "h0.00,0.00,0.00", true);
    }

  lightHandler();
  }

  
  yield();
}

void hsi2rgbw(float H, float S, float I, int* rgbw) {
  int r, g, b, w;
  float cos_h, cos_1047_h;
  H = fmod(H,360); // cycle H around to 0-360 degrees
  H = 3.14159*H/(float)180; // Convert to radians.
  S = S>0?(S<1?S:1):0; // clamp S and I to interval [0,1]
  I = I>0?(I<1?I:1):0;
  
  if(H < 2.09439) {
    cos_h = cos(H);
    cos_1047_h = cos(1.047196667-H);
    r = S*255*I/3*(1+cos_h/cos_1047_h);
    g = S*255*I/3*(1+(1-cos_h/cos_1047_h));
    b = 0;
    w = 255*(1-S)*I;
  } else if(H < 4.188787) {
    H = H - 2.09439;
    cos_h = cos(H);
    cos_1047_h = cos(1.047196667-H);
    g = S*255*I/3*(1+cos_h/cos_1047_h);
    b = S*255*I/3*(1+(1-cos_h/cos_1047_h));
    r = 0;
    w = 255*(1-S)*I;
  } else {
    H = H - 4.188787;
    cos_h = cos(H);
    cos_1047_h = cos(1.047196667-H);
    b = S*255*I/3*(1+cos_h/cos_1047_h);
    r = S*255*I/3*(1+(1-cos_h/cos_1047_h));
    g = 0;
    w = 255*(1-S)*I;
  }
  
  rgbw[0]=r;
  rgbw[1]=g;
  rgbw[2]=b;
  rgbw[3]=w;
}

void lightHandler(){
  //new and current lightStates
  static lightState newState;
  static lightState currentState;

  lightUpdater(&newState, currentState);
  lightChanger(newState, &currentState);
}

int lightChanger(lightState newState, lightState *currentState){
  
  currentState->red =newState.red;
  currentState->green =newState.green;
  currentState->blue =newState.blue;
  currentState->white =newState.white;
  //analogWrite(redPin, currentState->red);
  //analogWrite(greenPin, currentState->green);
  //analogWrite(bluePin, currentState->blue);
  _my9291.setColor((my9291_color_t) { currentState->red, currentState->green, currentState->blue, currentState->white });
  _my9291.setState(true);
}

void lightUpdater (lightState *newState, lightState currentState){

  //calculate new vars only if there is a new command
  if (newCommand){

      //convert from HSB to RGB
      /*int newRGB[3];
      H2R_HSBtoRGBfloat(nextState.hue, nextState.saturation, nextState.brightness, newRGB);
      newState->red = newRGB[0];
      newState->green = newRGB[1];
      newState->blue = newRGB[2]; 
      */
      int rgbw[4];
      hsi2rgbw(nextState.hue, nextState.saturation, nextState.brightness, rgbw);
 newState->red = rgbw[0];
 newState->green = rgbw[1];
 newState->blue = rgbw[2];
 newState->white = rgbw[3];
}
}
//MQTT callback
void callback(char* topic, byte* payload, unsigned int length) {

  //convert topic to string to make it easier to work with
  String topicStr = topic; 

  char newPayload[40];
  memcpy(newPayload, payload, length);
  newPayload[length] = '\0';

  //handle HSB updates
  if(payload[0] == 'h'){
    nextState.hue = atof(&newPayload[1]);
    nextState.saturation = atof(&newPayload[7]);
    nextState.brightness = atof(&newPayload[13]);

    newCommand = true;
    
    }
//package up status message reply and send it back out to the status topic
  strcpy(statusString, newPayload);
  myESP.publish(statusTopic, statusString, true);
}

igrr, 2017-05-08:
Originally reported in https://github.com/esp8266/Arduino/issues/612.
Fixed in master branch.

closed 2017-05-08 by igrr

Supersimo88, 2017-05-08:
I've already seen it but it still does not work for me

igrr, 2017-05-08:
Which version of Arduino ESP8266 core are you using?

Supersimo88, 2017-05-08:
2.3.0

igrr, 2017-05-08:
As mentioned, this is only fixed in master branch, not in any release, which explains why you are having this issue in 2.3.0. As a workaround, you can use git version of the core (instructions given in README.md).

Supersimo88, 2017-05-08:
I think I've done something wrong because I now get this error

exec: "D:\\Arduino\\arduino-1.8.1\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++": file does not exist

igrr, 2017-05-08:
Probable forgot to run `get.py`?

Supersimo88, 2017-05-08:
yes... i'm so stupid.
Compiled succesfully
Thanks a lot you made my day!

jack23233, 2017-07-06:
Having the same #612 error. 
deleted Arduino 1.6.x and installed Arduino 1.8.3  today
Boards Manager 'removed' esp8266 2.1.0 and installed 2.3.0 today
the same error appears. e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

I am in a Win10 system (not linux) and am unable to execute the 'git' command, even after creating the ../hardware/esp8266com folder.

After adding the esp8266com folder, my compile error now also reports:
WARNING: Error loading hardware folder C:\Program Files\Arduino\hardware\esp8266com
  No valid hardware definitions found in folder esp8266com.
WARNING: Error loading hardware folder C:\Program Files\Arduino\hardware\esp8266com
  No valid hardware definitions found in folder esp8266com.

I think thats understandable given that I could not execute a 'git' command.
I have downloaded a new Arduino-master.zip, but am reluctant to 'unzip' it .

other .ino files that do not use "exotic math" code will compile, download and run to the nodemcu 1.0 esp8266 device.

It seems that the Board Manager does not load the appropriate files into 
C:\Program Files\Arduino\hardware\esp8266com

I would really appreciate some help resolving this.