esp8266/Arduino #3218: WifiClient.write issue with long packets (> 2048 bytes)

fededim, 2017-05-09:
Hi,

I am trying to return an image (previously saved into SPIFFS) to a wifi client and I have this strange issue: if I send the image in chunks of maximum 2048 bytes everything works, instead if I send in chunks of 4096 bytes the data gets garbled. Basically I am:

1) Allocating a buffer
2) Reading file into the buffer
3) Writing buffer to wificlient

I checked if it was a read error by dumping some bytes of the read buffer, but the reads are fine as you can see from the two logs (one for 2048 chunks and one for 4096, _bytes dumped in i-eth send of 4096 buffer are equal to those in (2*i)-eth send of 2048 buffer_). **So there must be some problem with wificlient.write. Do you know if there is a maximum number of bytes which can be sent ?** I have tried also with client.flush after a write, but it does not work.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12F?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?NodeMCU?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
	  char resp[512],*buf;

          File f = SPIFFS.open(path, "r");

          sprintf(resp,"HTTP/1.1 200 OK\nServer: EspOven (Esp8266)\nContent-Length: %d\nConnection: Closed\nContent-Type: image/jpeg;\n\n",f.size());
          client.print(resp);

          buf=(char *) os_malloc(4096);
          
          int k=0;
          while (f.available() > 0) {
            int i = f.readBytes(buf, 4096);
            Serial.printf("EspOven: read %d bytes to client (%d): %02x %02x %02x %02x %02x %02x %02x %02x\n",i,k,buf[0],buf[1],buf[2],buf[3],buf[4],buf[5],buf[6],buf[7]);
            
            client.write((const uint8_t *) buf, i);
            client.flush();
            Serial.printf("EspOven: written %d bytes to client (%d)\n",i,k++);
            delay(1); // pass control to network stack
          }

          os_free(buf);
          
          f.close();

```

### Debug Messages

```
[b]Log 2048 chunks:
[/b]EspOven: Client connected from xxx.yy.zz.www:8917
EspOven: received request:
GET /20150918_235057.jpg HTTP/1.1
EspOven: method GET path /20150918_235057.jpg ver HTTP/1.1 freeheap 43200
EspOven: path /20150918_235057.jpg found, file len 192645 mime image/jpeg
[... 275 lines cut ...]
EspOven: written 4096 bytes to client (38)
EspOven: read 4096 bytes to client (39): d7 8a 9d 64 bf 96 26 c3
EspOven: written 4096 bytes to client (39)
EspOven: read 4096 bytes to client (40): 9a 92 45 c6 33 9e bd 68
EspOven: written 4096 bytes to client (40)
EspOven: read 4096 bytes to client (41): bf 95 2a 8d d8 dc 0f af
EspOven: written 4096 bytes to client (41)
EspOven: read 4096 bytes to client (42): f5 ef 54 04 32 1d c9 f7
EspOven: written 4096 bytes to client (42)
EspOven: read 4096 bytes to client (43): e9 42 86 1d 76 ee e3 d4
EspOven: written 4096 bytes to client (43)
EspOven: read 4096 bytes to client (44): 47 cd d3 27 ad 2a 80 cc
EspOven: written 4096 bytes to client (44)
EspOven: read 4096 bytes to client (45): c2 b6 de db 78 3f 85 43
EspOven: written 4096 bytes to client (45)
EspOven: read 4096 bytes to client (46): d7 24 da c4 e0 5c 95 cf
EspOven: written 4096 bytes to client (46)
EspOven: read 133 bytes to client (47): a6 7b 77 38 ef ed 9c fe
EspOven: written 133 bytes to client (47)
EspOven: Client xxx.yy.zz.www:8916 disconnecting
```

[image]
[image]
[image]

igrr, 2017-05-09:
Please clarify which version of the ESP8266 Arduino core you are using. If you have installed it using boards manager, open boards manager to see the version. If you have installed it using git, run `git describe` inside core directory to see the version. Thanks.

fededim, 2017-05-09:
Using arduino 1.8.1, installed version 2.3.0 through boards manager

[image]

fededim, 2017-05-09:
It should not be a problem with memory because ESP.getFreeHeap() reports 42k bytes free

igrr, 2017-05-09:
In version 2.3.0, `WiFiClient.write` will only write up to 2 * TCP_MSS, which is 2920 bytes. Return value of `client.write(...)` call will tell you how much was actually written. Then you can adjust your write pointer by the number of bytes which were written and perform the write operation again.

That being said, in the master branch this limitation has been removed, and WiFiClient::write will take care of sending *all* the supplied data, unless `WiFiClient::setNonBlocking` was called.

closed 2017-05-09 by igrr

fededim, 2017-05-09:
Ok thanks for the info, I imagine that the board manager version comes from the same branch, but it is based on an older base code, just to know when do you plan to update the board manager version to the newest main branch ? Meanwhile I did some test on performance also....it seems that network sending is slow, e.g. (inserted timestamp in logs) reading 2k from SPIFFS is about 1ms, while network sending takes between 20-40ms. Do you have any idea why ?

EspOven: received request:
GET /20150918_235057.jpg HTTP/1.1
EspOven: method GET path /20150918_235057.jpg ver HTTP/1.1 freeheap 42760
EspOven: path /20150918_235057.jpg found, file len 192645 mime image/jpeg
EspOven (13:53:51.406): read 2048 bytes to buffer (0): ff d8 ff e0 00 10 4a 46 freeheap 40024
EspOven (13:53:51.429): written 2048 bytes to client (0)
EspOven (13:53:51.430): read 2048 bytes to buffer (1): 00 00 00 00 00 00 00 00 freeheap 40024
EspOven (13:53:51.457): written 2048 bytes to client (1)
EspOven (13:53:51.459): read 2048 bytes to buffer (2): 00 00 00 00 00 00 00 00 freeheap 40024
EspOven (13:53:51.499): written 2048 bytes to client (2)
EspOven (13:53:51.500): read 2048 bytes to buffer (3): d1 cf d2 d0 ce d2 d1 d0 freeheap 40024
EspOven (13:53:51.538): written 2048 bytes to client (3)
EspOven (13:53:51.539): read 2048 bytes to buffer (4): b4 b3 b3 b2 b2 b4 b2 b2 freeheap 40024
EspOven (13:53:51.579): written 2048 bytes to client (4)
EspOven (13:53:51.580): read 2048 bytes to buffer (5): fd 31 ff 00 ec a9 8d e0 freeheap 40024
EspOven (13:53:51.596): written 2048 bytes to client (5)
EspOven (13:53:51.597): read 2048 bytes to buffer (6): 6d 70 3a 43 72 65 61 74 freeheap 40024
EspOven (13:53:51.617): written 2048 bytes to client (6)
EspOven (13:53:51.618): read 2048 bytes to buffer (7): 20 20 20 20 20 20 20 20 freeheap 40024
EspOven (13:53:51.636): written 2048 bytes to client (7)
EspOven (13:53:51.637): read 2048 bytes to buffer (8): 0a 20 20 20 20 20 20 20 freeheap 40024
EspOven (13:53:51.659): written 2048 bytes to client (8)
EspOven (13:53:51.661): read 2048 bytes to buffer (9): 20 20 20 20 20 20 20 20 freeheap 40024
EspOven (13:53:51.700): written 2048 bytes to client (9)

fededim, 2017-05-09:
Removed Boards manager version and updated to the latest from git, it works perfectly with 8192 chunks. Just the issue about speed, which is not enthusiastic, between 60 kb/s and 100 kb/s (I'm in the same room of the wifi ap), unluckily most of the time is spent inside WifiClient.write call.

EspOven: Client connected from xxx.yyy.zzz.www:10460
EspOven: received request:
GET /20150918_235057.jpg HTTP/1.1
EspOven: method GET path /20150918_235057.jpg ver HTTP/1.1 freeheap 39728
EspOven (15:15:32.698): path /20150918_235057.jpg found, file len 192645 mime image/jpeg
EspOven (15:15:32.702): read 8192 bytes to buffer (0): ff d8 ff e0 00 10 4a 46 freeheap 29248
EspOven (15:15:32.971): written 8192 bytes to client (0) freeheap 27648
EspOven (15:15:32.975): read 8192 bytes to buffer (1): b4 b3 b3 b2 b2 b4 b2 b2 freeheap 27648
EspOven (15:15:32.138): written 8192 bytes to client (1) freeheap 27648
EspOven (15:15:32.141): read 8192 bytes to buffer (2): 0a 20 20 20 20 20 20 20 freeheap 27648
EspOven (15:15:32.239): written 8192 bytes to client (2) freeheap 27648
EspOven (15:15:32.290): read 8192 bytes to buffer (3): 20 20 20 20 20 20 20 20 freeheap 29248
EspOven (15:15:32.363): written 8192 bytes to client (3) freeheap 26048
EspOven (15:15:32.366): read 8192 bytes to buffer (4): b4 53 d8 e4 b3 06 0b 9f freeheap 26048
EspOven (15:15:32.473): written 8192 bytes to client (4) freeheap 26048
EspOven (15:15:32.477): read 8192 bytes to buffer (5): d2 91 99 5c af 5f 98 fe freeheap 26048
EspOven (15:15:32.582): written 8192 bytes to client (5) freeheap 27648
EspOven (15:15:32.586): read 8192 bytes to buffer (6): 15 f8 84 a9 aa 6b 11 49 freeheap 27648
EspOven (15:15:33.710): written 8192 bytes to client (6) freeheap 27648
EspOven (15:15:33.715): read 8192 bytes to buffer (7): a4 89 b8 11 b9 5b 04 64 freeheap 27648
EspOven (15:15:33.813): written 8192 bytes to client (7) freeheap 26048
EspOven (15:15:33.817): read 8192 bytes to buffer (8): ce ad ab 6e 2a 57 76 df freeheap 26048
EspOven (15:15:33.910): written 8192 bytes to client (8) freeheap 27648
EspOven (15:15:33.914): read 8192 bytes to buffer (9): 28 f7 d9 97 19 5d 1d 7a freeheap 27648
EspOven (15:15:33.996): written 8192 bytes to client (9) freeheap 27648
EspOven (15:15:33.001): read 8192 bytes to buffer (10): 03 d6 30 58 63 ea 2a ca freeheap 27648
EspOven (15:15:33.094): written 8192 bytes to client (10) freeheap 29248
EspOven (15:15:33.098): read 8192 bytes to buffer (11): 5f 46 e3 24 67 a5 35 24 freeheap 29248
EspOven (15:15:33.187): written 8192 bytes to client (11) freeheap 27648
EspOven (15:15:33.191): read 8192 bytes to buffer (12): 1d be 5c 0e 69 b2 af 9c freeheap 27648
EspOven (15:15:33.284): written 8192 bytes to client (12) freeheap 27648
EspOven (15:15:33.288): read 8192 bytes to buffer (13): fe 9d 48 dc 10 60 75 e9 freeheap 27648
EspOven (15:15:33.388): written 8192 bytes to client (13) freeheap 29248
EspOven (15:15:33.393): read 8192 bytes to buffer (14): 04 a3 cc dc 0e 3d 71 8f freeheap 29248
EspOven (15:15:33.494): written 8192 bytes to client (14) freeheap 26048
EspOven (15:15:33.498): read 8192 bytes to buffer (15): fd d9 03 a9 e0 53 40 47 freeheap 26048
EspOven (15:15:33.632): written 8192 bytes to client (15) freeheap 26048
EspOven (15:15:33.636): read 8192 bytes to buffer (16): ed c1 fe 74 ef b3 c9 24 freeheap 26048
EspOven (15:15:34.743): written 8192 bytes to client (16) freeheap 27648
EspOven (15:15:34.747): read 8192 bytes to buffer (17): ee cc 40 05 b0 0e 0b 8a freeheap 27648
EspOven (15:15:34.872): written 8192 bytes to client (17) freeheap 27648
EspOven (15:15:34.878): read 8192 bytes to buffer (18): 71 eb 55 e3 fd d8 2b 9c freeheap 27648
EspOven (15:15:34.958): written 8192 bytes to client (18) freeheap 27648
EspOven (15:15:34.962): read 8192 bytes to buffer (19): af f9 fa 56 75 c0 c2 6d freeheap 27648
EspOven (15:15:34.050): written 8192 bytes to client (19) freeheap 29248
EspOven (15:15:34.054): read 8192 bytes to buffer (20): 9a 92 45 c6 33 9e bd 68 freeheap 29248
EspOven (15:15:34.150): written 8192 bytes to client (20) freeheap 26048
EspOven (15:15:34.154): read 8192 bytes to buffer (21): f5 ef 54 04 32 1d c9 f7 freeheap 26048
EspOven (15:15:34.282): written 8192 bytes to client (21) freeheap 27648
EspOven (15:15:34.287): read 8192 bytes to buffer (22): 47 cd d3 27 ad 2a 80 cc freeheap 27648
EspOven (15:15:34.379): written 8192 bytes to client (22) freeheap 27648
EspOven (15:15:34.382): read 4229 bytes to buffer (23): d7 24 da c4 e0 5c 95 cf freeheap 27648
EspOven (15:15:34.451): written 4229 bytes to client (23) freeheap 29248
EspOven (15:15:34.452): transferred 192645 bytes to client, speed 69.00 KB/s
EspOven: Client xxx.yyy.zzz.www:10460 disconnecting

igrr, 2017-05-09:
100kB/sec sounds reasonable, you can expect the data rate to be 2*TCP_MSS/roundtrip_time, where roundtrip_time can be estimated using ping.

You can possibly get some small improvement if you don't fragment the file yourself but pass it as a `Stream` to WifiClient​.write.