esp8266/Arduino #3221: vsnprintf => ESP crashes on start

Adminius, 2017-05-09:
### Basic Infos

#### Hardware
Hardware:			ESP-12F on WeMos D1 mini R2
Core Version:      	2.3.0

### Description

i have a problem with a vsnprintf function.
I use this https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.h debug-lib
ESP crashes on start and only stack error is shown.
This lib works fine on 328p, 32u4 and SAMD21...

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz

### Debug Messages

Stack and decode:
https://gist.github.com/Adminius/4a08ac05888eb674dd5c11598214be49

### Sketch

```
//https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.h
//https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.cpp
#include <DebugUtil.h> 

// ESP8266 use the 2nd serial port with TX only (GPIO2 / D4)
#define DEBUGSERIAL Serial1

void setup() {
  DEBUGSERIAL.begin(9600);
  Debug.setPrintStream(&DEBUGSERIAL);
  Debug.println(F("Test"));
}

void loop() {

}

gerrylenz, 2017-05-09:
void setup() {
  DEBUGSERIAL.begin(9600);
  Debug.setPrintStream(&DEBUGSERIAL);
  Debug.println(F("Test"));
}

 

Change to

 

void setup() {
  DEBUGSERIAL.begin(9600);
  DEBUGSERIAL.setPrintStream(&DEBUGSERIAL);
  DEBUGSERIAL.println(F("Test"));
}

 

Gerry

 

https://gist.github.com/Adminius/4a08ac05888eb674dd5c11598214be49

//https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.h
//https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.cpp

Adminius, 2017-05-09:
this is not a reason... and solves not my problem:
```
ESP_crash:10: error: 'class HardwareSerial' has no member named 'setPrintStream'

   DEBUGSERIAL.setPrintStream(&DEBUGSERIAL);

               ^

exit status 1
'class HardwareSerial' has no member named 'setPrintStream'

Adminius, 2017-05-12:
@igrr is a release date of 2.4.0 already known? or do you have any temporarry solution for my problem?

igrr, 2017-05-12:
No idea, TBH. See list of things to be done for 2.4.0 here: https://github.com/esp8266/Arduino/projects/1

igrr, 2017-05-12:
Oh, this should actually be already fixed in git version. If you can follow "install from git" instructions in README.md, that would be the workaround.

Adminius, 2017-05-23:
i've tried 2.4.0-rc1... 
the same bug as with 2.3.0

pieman64, 2017-05-23:
@Adminius git version not rc1 version.

Adminius, 2017-05-23:
i can't compile python on win10...

igrr, 2017-06-05:
That's the bug in KonnektingDeviceLibrary, it attempts to use `vsnprintf` with a format string defined in PROGMEM, if the architecture is not AVR:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/src/DebugUtil.cpp#L46-L50

If you change these lines to use `vsnprintf_P` on the ESP8266 as well, the code will work.

closed 2017-06-05 by igrr