esp8266/Arduino #3224: Espcomm_sync failure Wemos D1R2

johnwa27, 2017-05-10:
I'm getting this well known failure.  All the responses seem to place blame on the users code or wiring. But it's happening to me on a Wemos board with nothing connected to it.    It happens randomly, sometimes my code uploads, other times it will take 3 or 4 tries.  Closing the serial monitor seems to help a bit.  

This is not an issue with my code, my code works when it uploads and it happens with several different working programs.
It's a hardware/ide issue.  It's made even worse since it has to recompile every upload attempt.

When is someone going to take this issue seriously?

davisonja, 2017-05-10:
I think it's fair to say that it is taken seriously, but it has a myriad of
potential causes meaning that there isn't one simple fix for everyone.

For example, I've got a "Wemos D1R2" board (I'm sure it's a clone, but
still) which has a flaky USB socket. It needs to be at just the right angle
in order for the board to appear on my computer as a serial port - the
symptoms, if the USB cable isn't just right are the 'espcomm_sync' errors.
Here it's an easy fault - the USB connector on the board isn't attached
properly, requiring pressure from the connected USB cable to complete the
connection.
Of course, it's a localised hardware fault to that board, so it's not
something that anyone else can either fix, or diagnose for me.

If you've got the same kind of issue you might need to talk to your
supplier.

igrr, 2017-05-10:
This issue was moved to igrr/esptool-ck#52

closed 2017-05-10 by igrr