esp8266/Arduino #3225: Current CHUNKED implemantion in WebSever GitHub version not complete + HOW TO FIX

horendus, 2017-05-10:
In testing the CHUNKED html send implementation in the current GitHub version I have found that it is not complete because it does not send a 0bytes body content message to the client to tell the browser to TERMINATE the chunked transfer session. 

This causes the browser connection to remain active for an extra 2 seconds or so even though the entire HTML chunked transfer is complete This slows down the browser!

In looking at FireFox dev tools a chunked transfer of 5*3000byte HTML pieces is taking over 2300ms to complete with the later 2 seconds the browser is just sitting there waiting for its own transfer time out to kicks in.

After modifying the Webserver library I can have the transfer/transaction complete in just 150ms.

ADD to ESP8266WebSwerver.cpp
```
void ESP8266WebServer::endChunkedContent() {
	const char * end = "0\r\n\r\n";
	_currentClient.write(end, 5);
}
```

ADD to ESP8266WebServer.h
`void endChunkedContent();   //Call this last when doing multiple server.sends (in CHUNKED trasnfers)`

Then to do proper CHUNKED html transfers using the following method

```
  server.setContentLength(CONTENT_LENGTH_UNKNOWN);   //Enable Chunked Transfer
  server.send(200, "text/plain", webPagePart1());                      //Send first part WITH header
  server.sendContent(webPagePart2());                                    //Send next chunk of html
  server.sendContent(webPagePart3());                                    //Send next chunk of html
  server.endChunkedContent();                                               //Tell browser no more content is coming
```

vortigont, 2017-05-23:
server.sendContent("");
server.client().stop()
can do the trick to send empty chunk and close the connection fast.

bryceschober, 2017-07-27:
@igrr: Is the proposed fix not good enough, or are you planning on doing something different?

May i suggest an interface more like `server.finish()` that we could used to search-and-replace all the examples currently doing `server.client().stop()` with zero-length responses or content-length-unknown responses. It could properly terminate when chunking, and call server.client.stop() regardless, along the lines of:
```c++
void ESP8266WebServer::finish() {
  if(_chunked) {
    sendContent("");
  }
  _currentClient.stop();
}
```

igrr, 2017-12-14:
@bryceschober If this issue is still relevant, could you please have a look at the proposed fix in #3963? It should be compatible with existing sketches which use chunked encoding, i think.

closed 2017-12-26 by igrr