esp8266/Arduino #3226: MISSING 2MB SPIFFS option for NodeMCU which allows OTA UPDATES > 500KB~ + FIX

horendus, 2017-05-10:
I found that sketches over 500KB wont OTA Update on the 3MB SPIFFS version as there is not enough space before the SPIFFS memory allocation.

1MB SPIFFS seems like a waste, so we need a 2MB SPIFFS option that has 1MB BEFORE the SPIFFS flash allocations to allow > 500KB  OTA updates

Bellow is how I achieve this on my NodeMCU (actually a custom board but I still use nodeMCU board def.)

Add this to AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\board.txt file 
Scroll down to NodeMCU 1.0v and add it there. 

```
nodemcuv2.menu.FlashSize.4M2M=4M (2M SPIFFS)
nodemcuv2.menu.FlashSize.4M2M.build.flash_size=4M
nodemcuv2.menu.FlashSize.4M2M.build.flash_ld=eagle.flash.4m2m.ld
nodemcuv2.menu.FlashSize.4M2M.build.spiffs_start=0x200000
nodemcuv2.menu.FlashSize.4M2M.build.spiffs_end=0x3FB000
nodemcuv2.menu.FlashSize.4M2M.build.spiffs_blocksize=8192
nodemcuv2.menu.FlashSize.4M2M.build.spiffs_pagesize=256
nodemcuv2.menu.FlashSize.4M2M.upload.maximum_size=1044464 
```

Add a LD file to AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld called eagle.flash.4m2m and insert the bellow contents. Hint. Just make a .txt file and add .ld file extension to the end

```
/* Flash Split for 4M chips */
/* sketch 1019KB */
/* empty  1024KB */
/* spiffs 2028KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xfeff0
}

PROVIDE ( _SPIFFS_start = 0x40400000 );
PROVIDE ( _SPIFFS_end = 0x405FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x2000 );

INCLUDE "../ld/eagle.app.v6.common.ld" 

```

igrr, 2017-05-10:
Thanks for the submission. Please consider posting your fixes and additions in Pull Requests :)

horendus, 2017-05-10:
No worries will do.

devyte, 2017-05-10:
I don't really need it, but I have been wondering about why there wasn't a
2M option for the longest time. Your PR will soothe my OCD :)

or mute
the thread

capedra, 2018-03-03:
@igrr @ivankravets 
Friends, please, add support to 4m2m LD Script. I don't know how to open a PR. It's too confusing. 😕
Best Regards!