esp8266/Arduino #3227: GitHub vs 2.3.0 Release Ram Usage > Stable 2.3.0 uses much less RAM?

horendus, 2017-05-10:
Just noticed that compiling my (large) firmware with the  2.3.0 Stable Release vs the current GitHub version has a large RAM usages difference after boot.

Current GitHub:    11136 remaining
2.3.0 Stable:          13904 remaining

These are the (ESP8266) libraries being used.
#include <WiFiClient.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266WebServer.h>
#include <ESP8266httpUpdate.h>

Any idea where all the extra RAM usage could be coming from?

igrr, 2017-05-10:
Do you happen to use TLS (WiFiClientSecure) or httpUpdate over HTTPS in your sketch?

If you upload both elf files somewhere and post a link to them, i can probably tell more.

horendus, 2017-05-10:
The free size is reported by ESP.getFreeHeap() but I can also see the change of RAM usage in the Arduio DIE output.

Things being initalised from ESP libraries
MDNS.begin();
MDNS.addService();

horendus, 2017-05-10:
(sorry) Where do I retrieve the ELF files from?

igrr, 2017-05-10:
If you enable verbose compilation output in Arduino IDE Preferences, you will get the path to the compiled .bin file in the compilation log. The ELF file is going to be in the same directory, with an .elf extension.

devyte, 2017-09-06:
@horendus will you be posting the elf files? Also, please check heap at the very beginning of setup to know at value it starts with.

igrr, 2017-12-28:
No feedback for a while, closing.

closed 2017-12-28 by igrr